geopm/geopm #11: cpu-frequency-platform

sssylvester, 2016-05-17:
To enable deciders that use CPU frequency for control need to
implement a Platform plugin with right granularity of
measurement and control.

eastep, 2016-11-14:
I propose inclusion in v0.3.0

sssylvester, 2017-05-23:
The Decider update_policy() method will set RAPL limits in walk_down() evenly using the base class method (i.e. no telemetry used to inform choice).  It will dynamically set frequency targets in walk_up().  Platform needs to be updated to deal with those two different policy vectors, and the Policy class needs to support a map between control type (power/frequency) and the vector of policy values.  Note that the vectors for power policy and frequency policy will not be of equal length, power policy vector is shorter than the frequency policy vector.

cmcantalupo, 2017-12-11:
This work is partially done, and the parts that are not complete have other issues posted:

https://github.com/geopm/geopm/issues/128
https://github.com/geopm/geopm/issues/127
https://github.com/geopm/geopm/issues/130

closed 2017-12-11 by cmcantalupo