geopm/geopm #39: Tutorial_0 fails to run on Sandy Bridge

amarathe84, 2016-09-12:
I'm trying to run tutorial_0 on a Sandy Bridge cluster. I get the following error when I submit a job for tutorial_0 through the msub script listed below: 

**MSUB script:**
```
#!/bin/bash
#MSUB -l nodes=2
#MSUB -l partition=cab
#MSUB -q pbatch
#MSUB -V
#MSUB -l walltime=00:05:00
#MSUB -o tutorial_0.out
#MSUB -e tutorial_0.err

#Following setup already added in ~/.bashrc
#use mvapich2-gnu-2.1
#use gcc-4.9.2p
#use ruby-2.2.4
#export PATH=$(HOME)/geopm/install-cab/bin:$PATH
#export LD_LIBRARY_PATH=$(HOME)/geopm/install-cab/lib:$LD_LIBRARY_PATH

srun -N 2 -n 4 --auto-affinity=start=0,cpt=1 $(HOME)/geopm/geopm-cab/tutorial/tutorial_0.sh

```
**Error:** 
```
Error: <geopm> Could not write to MSR device: MSR value to be written was modified by the mask! Desired = 0x2aaaacdd0db8 After mask = 0xacdd0db8: at geopm/src/PlatformImp.cpp:228
Error: <geopm> Could not write to MSR device: MSR value to be written was modified by the mask! Desired = 0x2aaaacdd0db8 After mask = 0xacdd0db8: at geopm/src/PlatformImp.cpp:228
Warning: <geopm> Controller not found, running without geopm.
Warning: <geopm> Controller not found, running without geopm.
srun: error: cab1179: task 0: Exited with exit code 241
srun: error: cab1180: task 2: Exited with exit code 241
```

The msub file lists the environment variables set through ~/.bashrc at shell init. Tutorial_0.sh file has the following contents: 

```
source tutorial_env.sh

LD_PRELOAD=$GEOPM_LIBDIR/libgeopm.so \
LD_DYNAMIC_WEAK=true \
GEOPM_PMPI_CTL=process \
GEOPM_REPORT=tutorial_0_report \
GEOPM_TRACE=tutorial_0_trace \
GEOPM_SHMEM_KEY=geopm-shm-3-`whoami` \
./tutorial_0
```

I also noticed that 'geopmpolicy -w' shows related output so I'm including it here: 

```
$ geopmpolicy -w
Error: <geopm> Could not write to MSR device: MSR value to be written was modified by the mask! Desired = 0x2aaaab05ddb8 After mask = 0xab05ddb8: at geopm/src/PlatformImp.cpp:228
Error: <geopm> Could not write to MSR device.

```

cmcantalupo, 2016-09-16:
This is really a duplicate of issue number 1:

https://github.com/geopm/geopm/issues/1

It does show how unhelpful geopmpolicy -w is for someone trying to create a white list without root permission.

cmcantalupo, 2016-10-31:
Not being able to run
    geopmpolicy -w
without root is frustrating, but the real problem here is that the write mask for the frequency control MSR is not correct in the msr-safe repo.  This issue is resolved by this pull request:

https://github.com/LLNL/msr-safe/pull/6

@amarathe84 any way you can help to get this pull request merged is appreciated.

closed 2016-11-30 by cmcantalupo