geopm/geopm #41: Unlocking RAPL MSR on Dell servers

etano, 2016-09-21:
I'm trying to do power limiting on a Dell server (R630), and the RAPL MSR lock bit is set. I can't find a configuration of BIOS settings to unlock it.

Has anyone else experienced the same? Any suggestions?

cmcantalupo, 2016-09-21:
I've been asking around about this one.  The best advice I have so far is to look into the "System Profile" and "Performance Profile" menus in the Dell BIOS.  Unfortunately I do not have a specific BIOS setting to suggest.  I'll be sure to post here if I get any more information.

etano, 2016-09-21:
Certainly, unfortunately there's nothing explicit there about power limiting/locks. Dell has their own power management, which I've tried disabling, to no avail. Do you have a Dell technical contact that could help?

etano, 2016-10-17:
Just an update on this. All current Dell BIOS have the MSR RAPL lock bit set.

Inquiring now about making a custom BIOS.

cmcantalupo, 2016-10-31:
Thanks for the update.  A custom BIOS sounds messy.  Wish we could be more helpful.  It is discouraging that Dell has made this design decision.