geopm/geopm #43: tutorial 1 count reporting incorrect

cmcantalupo, 2016-09-23:
The reports files from tutorial 1 intermittently report incorrect count values.  In the case below, the outer-sync count appears to be random memory (uninitialized or corrupted), and the mpi-sync count is 19 instead of 20.  This was reproduced with SHA 8df243b.

Reports from tutorial 1 run on two nodes:

##### geopm 0.1.1+dev248g8df243b #####

Profile: ./tutorial_1
Region mpi-sync:
        runtime (sec): 0
        energy (joules): 0
        frequency (%): 0
        count: 19
Region outer-sync:
        runtime (sec): 85.0606
        energy (joules): 11733.6
        frequency (%): 106.289
        count: 1.84467e+19
##### geopm 0.1.1+dev248g8df243b #####

Profile: ./tutorial_1
Region mpi-sync:
        runtime (sec): 0.18275
        energy (joules): 21.3884
        frequency (%): 107.692
        count: 20
Region outer-sync:
        runtime (sec): 85.0312
        energy (joules): 11280.3
        frequency (%): 106.254
        count: 0

sssylvester, 2016-10-12:
I have traced this back to commit 481ac6bb5754602d06929c424b490357c2b159b3.
Currently looking for a solution.

sssylvester, 2016-10-12:
It is possible I am chasing a new bug and the behavior is not related. Looking into it.

sssylvester, 2016-10-13:
This was indeed a different bug showing similar behavior. After fixing that one, this bug still reproduces.

sssylvester, 2016-11-09:
This problem is fixed/invalid as of bdbfd22611253a6132cc41d88aa5ed3c2eb28fad.
We have changed the way MPI time is accounted for and reported. The tutorial logs are now correct.

closed 2016-11-09 by sssylvester