geopm/geopm #83: MPI_Gather and geopm

slabasan, 2017-01-16:
I'm running into troubles doing a simple MPI_Gather of rank IDs when using geopm's profiling feature. The number of ranks is correct, but the results are garbage values. I'm running my example with 4 nodes and 8 tasks, 2 tasks per node. With geopm, I would expect the rank IDs to be 0 and 1. 

I've put a reproducer code here: 
git clone https://github.com/slabasan/mpigather-geopm.git

Let me know if you have any questions.

cmcantalupo, 2017-01-17:
Thanks for finding this bug and for the nice reproducer!  I think the fix is here:

https://github.com/cmcantalupo/geopm/tree/cmcantalupo-fix-gather-wrapper

Let me  know if that works for you.  We had forgotten to swap the communicator for the MPI_Gather() and MPI_Gatherv() wrapper routines.

closed 2017-01-17 by cmcantalupo

slabasan, 2017-01-17:
@cmcantalupo Thank you! Problem fixed!

commit 0047506f (2017-01-17): Fix missing comm swap call in MPI_Gather() and MPI_Gatherv() wrappers.

- In github issues, fixes #83.

Change-Id: Id7c4173c783bd7d5490fc00fb37f93c4073180fd