geopm/geopm #84: Governing Plugin Tutorial 3

slabasan, 2017-01-21:
When I run tutorial 3 in a similar fashion to the online video, I'm seeing a different observed node power budget from GEO's telemetry data when using the Governing Plugin. My understanding of this policy is that the power caps of the RAPL domains will be toggled such that each node draws no more than the specified budget.

I ran four experiments of Tutorial 3 with 100 iterations using two nodes, 5 tasks per node on Catalyst. I used a node power budget of 150W, 180W, 200W, and 300W (an initial thought was that 150W was too low of a node power budget given the minimum power of our processor). I manually computed the effective node power draw from the PKG and DRAM data collected, and plotted a histogram of the full range of power values as well as a subset of the power values centered around the specified bound. Despite the specified node power budget, I'm seeing that the majority of values exceed the bound by ~30W (vertical red line). I also notice that as the node power budget is increased, the majority of samples fall into a higher and higher power range, so it seems that the plugin is keeping the node to a budget.

I collected info from the Governing Plugin..the decisions on the RAPL power caps do add to the specified node power bound. I'm in the process of checking the values written to the power limit MSRs. Is this a known behavior of the Governing Plugin or am I misunderstanding how to read the data?

Thanks again! Let me know if you have any questions.

[tutorial3_150W_hist.pdf](https://github.com/geopm/geopm/files/721491/tutorial3_150W_hist.pdf)
[tutorial3_180W_hist.pdf](https://github.com/geopm/geopm/files/721495/tutorial3_180W_hist.pdf)
[tutorial3_200W_hist.pdf](https://github.com/geopm/geopm/files/721490/tutorial3_200W_hist.pdf)
[tutorial3_150W_hist.pdf](https://github.com/geopm/geopm/files/721491/tutorial3_150W_hist.pdf)

cmcantalupo, 2017-01-23:
@slabasan 
Thanks for the feedback Stephanie!  It certainly seems like something is not going as expected here.  It looks like you posted the 150W data twice and are missing the 300W data.  I'll start digging into this soon and try to reproduce on catalyst.

slabasan, 2017-01-23:
@cmcantalupo: I've updated the 300W data. If you'd like the raw data, let me know. I ended up reducing the number iterations in Tutorial 3 from 500 to 100 for time sake. 500 iterations seemed to take much longer on Catalyst than on your Mr. Fusion system. Thanks!

cmcantalupo, 2017-01-28:
We've been looking into this issue and are able to reproduce.  The energy data in the trace is pretty noisy and we are wondering if inaccuracies in the power calculation based on differencing noisy energy readings is part of the issue.  I switch from a 2 point stencil to a 5 point stencil for estimating the derivative of energy with respect to time here:

https://github.com/cmcantalupo/geopm/tree/cmcantal-five-point-derivative

Even with this patch, we are still seeing similar inability of the governor to keep the processor within the power limit.

slabasan, 2017-01-28:
Thanks for the update. I was also thinking the sampling frequency of 1 ms is too high for this platform, also leading to noisy energy readings. Is there a simple way to reduce the frequency of telemetry collection?

cmcantalupo, 2017-01-28:
@sssylvester enabled the independent sample rate feature which is set here:

https://github.com/geopm/geopm/blob/dev/src/Controller.cpp#L187

The sample rate is not currently tied to a method returned by the Platform, it is simply hard coded in the Controller constructor.  This should be changed so that there is a Platform::sample_period_limit() method which the Controller uses to set it's local member (m_sample_rate_limit member is a sample period in units of seconds, not a rate).  In the mean time, you can increase m_sample_rate_limit which holds the sample period as is used here:

https://github.com/geopm/geopm/blob/dev/src/Controller.cpp#L515

Hopefully this improves the reliability of the GoverningDecider.

slabasan, 2017-01-30:
I increased the sampling rate to 100 ms and 500 ms, and the seconds reported in the trace appear inconsistent. If I take the difference in seconds, I'm seeing zero at times, which is producing NANs when computing the power consumption. I've found some pairs of samples (trace below using 500 ms sampling frequency) which have identical values, except they differ in the progress.0 and progress.1. Example pairs are lines (11, 12) and (15, 16). Looking at earlier traces with a sampling frequency of 1 ms, I don't see this problem exposed. I did some debugging in the sample period mentioned above, and those look sane.

[tutorial_3_governed-foo_trace-catalyst321.txt](https://github.com/geopm/geopm/files/739904/tutorial_3_governed-foo_trace-catalyst321.txt)

cmcantalupo, 2017-01-31:
It is expected that some of the data in the trace output will have the same time stamp as nearby samples.  This occurs precisely when a region is entered or exited (progress == 0 or progress == 1).  Also, note that if the sample period is increased, then the control period must be at least as big m_control_rate_limit:

https://github.com/geopm/geopm/blob/dev/src/Controller.cpp#L297

We are still doing some experiments to figure out what rates are best for each architecture, and trying to debug some issues around the GoverningDecider's subtraction of DRAM power usage from the budget.  We will post back here soon when we get to the bottom of it all.  Thanks!

cmcantalupo, 2017-02-04:
We are making some adjustments to the GoverningDecider.  A number of improvements were made to other parts of the code along the way while debugging this issue: more accurate RAPL window setting, more accurate power calculation from energy measurements, and platform control of sample rate.  The root of the problem, however, seems to be in the GoverningDecider::update_policy() method which has been rewritten.  We will be posting the fixes and improvements soon.

slabasan, 2017-02-04:
Thank you! Looking forward to seeing the changes.

cmcantalupo, 2017-02-14:
We just merged a number of changes into dev including https://github.com/geopm/geopm/commit/95122371d7edaf29b75cf58b2601c8edb25f32ee which rewrote the governing decider.  Any feedback about the current state of the tip of dev is appreciated.  These changes should fix the over budget issues you were seeing.  We are in the process of doing extensive integration tests on the current tip of the dev branch.

slabasan, 2017-02-15:
I will try and circle back to this by the end of the week.

cmcantalupo, 2017-02-18:
@slabasan

We are testing out an alternative option for setting the sample rate by checking for power counter updates.  This change is posted here:

https://github.com/cmcantalupo/geopm/tree/cmcantalupo-platform-rate-ctl

it is under review here:

https://review.gerrithub.io/#/c/348992/

cmcantalupo, 2017-03-10:
I think we have everything nailed down with this issue.

closed 2017-03-10 by cmcantalupo

slabasan, 2017-03-14:
The data is looking good to me as well. Thanks again!