geopm/geopm #110: Build fails on Quartz (TOSS 3)

amarathe84, 2017-10-20:
Master build fails at the step to build dummyf90.f90 on Quartz (TOSS 3):

...
2.2-intel-17.0.0/include  -c -o test/fortran/libmtest_la-dummyf90.lo `test -f 'test/fortran/dummyf90.f90' || echo './'`test/fortran/dummyf90.f90
libtool: compile:  gfortran -c -fopenmp -msse4.2 -fno-range-check -I/usr/tce/packages/mvapich2/mvapich2-2.2-intel-17.0.0/include -I/usr/tce/packages/mvapich2/mvapich2-2.2-intel-17.0.0/include -c test/fortran/dummyf90.f90  -fPIC -o test/fortran/.libs/libmtest_la-dummyf90.o
Fatal Error: Reading module mpi at line 1 column 2: Unexpected EOF
make[1]: *** [test/fortran/libmtest_la-dummyf90.lo] Error 1
make[1]: Leaving directory `/g/g92/marathe1/quartz/geopmtest'
make: *** [all] Error 2

Commenting out the following in 'am__objects_1' temporarily fixes the issue: 
test/fortran/libmtest_la-mtestf90.lo
test/fortran/libmtest_la-dummyf90.lo

I've the following modules loaded in my environment:
module load intel/17.0.0 1>&2 > /dev/null
module load mvapich2/2.2 1>&2 > /dev/null

tpatki, 2017-10-20:
> Commenting out the following in 'am__objects_1' temporarily fixes the issue:
test/fortran/libmtest_la-mtestf90.lo
test/fortran/libmtest_la-dummyf90.lo

This works for "make", but when I do a "make check", it fail again with a similar error on being unable to read mpi module/unexpected EOF on a different file.

```
gfortran -fopenmp -msse4.2 -fno-range-check   -I/usr/tce/packages/mvapich2/mvapich2-2.2-intel-17.0.0/include -I/usr/tce/packages/mvapich2/mvapich2-2.2-intel-17.0.0/include  -c -o test/fortran/test_fortran_allctypesf90-allctypesf90.o `test -f 'test/fortran/allctypesf90.f90' || echo './'`test/fortran/allctypesf90.f90
Fatal Error: Reading module mpi at line 1 column 2: Unexpected EOF
make[2]: *** [test/fortran/test_fortran_allctypesf90-allctypesf90.o] Error 1
make[2]: Leaving directory `/g/g90/patki1/src/ecprt/geopm'
make[1]: *** [check-am] Error 2
make[1]: Leaving directory `/g/g90/patki1/src/ecprt/geopm'
make: *** [check] Error 2
```

cmcantalupo, 2017-10-21:
If disabling fortran is an option for you this can be worked around by configuring without fortran support:

    ./configure --disable-fortran ...

The fortran tests for the MPI wrappers fail unless run against mvapich2.  We should probably just remove these fortran tests because they are MPI implementation specific.  Thanks for the feedback.

cmcantalupo, 2017-10-21:
A simple work around for disabling the fortran tests without disabling the fortran build is here:

https://github.com/cmcantalupo/geopm/commit/9eb82b46d52ec53ca432e6695033f01f74e4f9e0

maiterth, 2017-10-25:
This is an issue regarding the compiler used. You should not use gfortran but mpifort /mpif77.
This is working on the system:
./configure  CC=mpicc CXX=mpicxx F77=mpif77 FC=mpiifort MPIFC=mpiifort MPIF77=mpif77

cmcantalupo, 2017-10-28:
Proposal to remove the MVAPICH2 fortran tests is here:

https://review.gerrithub.io/#/c/384000/

Feel free to comment.

bgeltz, 2017-11-14:
Chris's patch has been merged.  The mvapich2 tests that were causing this issue have been removed.

closed 2017-11-14 by bgeltz