geopm/geopm #111: [GEOPM - Tutorials] Issues while running the tutorials

matsKNY, 2017-10-25:
Hello,

I am creating this issue concerning the tutorials because I failed to make them work. I will try to be thorough, so this issue might be a little long.

Some context first:

- I pulled the github repository and followed the README to compile GEOPM and the tutorials from the sources. Except the fact that I had to modify the file **configure.ac** to delete the **-Werror** compilation flag, everything went well (in **test/fortran/attrlangc.c**, some deprecated MPI functions are called, among which **MPI_Keyval_create**. I checked the reference guide of MPI, it is deprecated since MPI 2.0);

- The cluster I am working with has heterogeneous computes nodes. All of them have two sockets, embedding two identical processors from the Haswell family (either two _Intel Xeon E5-2670 v3_ or two _Intel Xeon E5-2650 v3_). The interconnect is an InfiniBand EDR fabric, and the nodes are running RHEL7.4. The **intel_pstate** CPU frequency driver is disabled, the cpufreq governor is set to **performance**. The MSR driver is loaded, and the command `geopmpolicy -w` is working on the nodes I tried to run the tutorials onto (by the way, my user account on the nodes has root privileges).

What I tried:

- I tried to use **tutorials/tutorial_0.sh** to run the first tutorial thanks to **geopmsrun** (two nodes and ten tasks). Since the nodes are not homogeneous, I slightly modified the content of **tutorials/tutorial_0.sh** in order to give a nodelist of homogeneous nodes, namely mexico48 and mexico49.

```bash 
geopmsrun  -w mexico[48,49] \    
           -N 2 \  
           -n 8 \  
           --geopm-preload \  
           --geopm-ctl=process \  
           --geopm-report=tutorial_0_report \  
           --geopm-trace=tutorial_0_trace \  
           -- ./tutorial_0
```

And below is the output I am getting:
`<geopmpy.launcher> list index out of range`

I have looked into the source code of **geopmpy.launcher**, and it seems I have specified the nodelist as it should be. Any mistake ?

- Since the "simple" way seemed to be a dead-end, I tried to use **MPIEXEC**, almost just as @cmcantalupo did in the tutorial videos.

```bash
user@mexico48 $ export GEOPM_RM=NONE
user@mexico48 $ export MPIEXEC="srun --reservation=Power-GEOPM --exclusive --cpu_bind=v,mask_cpu:0x1,0x2,0x4,0x8,0x10 -N 2 -n 10"
user@mexico48 $ ./tutorial_0.sh
        cpu_bind=MASK - mexico48, task  1  1 [648]: mask 0x2 set
        cpu_bind=MASK - mexico48, task  4  4 [652]: mask 0x10 set
        cpu_bind=MASK - mexico48, task  0  0 [645]: mask 0x1 set
        cpu_bind=MASK - mexico48, task  2  2 [650]: mask 0x4 set
        cpu_bind=MASK - mexico48, task  3  3 [651]: mask 0x8 set
        cpu_bind=MASK - mexico49, task  8  3 [35418]: mask 0x8 set
        cpu_bind=MASK - mexico49, task  6  1 [35416]: mask 0x2 set
        cpu_bind=MASK - mexico49, task  7  2 [35417]: mask 0x4 set
        cpu_bind=MASK - mexico49, task  5  0 [35415]: mask 0x1 set
        cpu_bind=MASK - mexico49, task  9  4 [35419]: mask 0x10 set
        [mexico48:4059] *** An error occurred in MPI_Comm_rank
        [mexico48:4059] *** reported by process [140303995502592,2]
        [mexico48:4059] *** on communicator MPI_COMM_WORLD
        [mexico48:4059] *** MPI_ERR_COMM: invalid communicator
        [mexico48:4059] *** MPI_ERRORS_ARE_FATAL (processes in this communicator will now abort,
        [mexico48:4059] ***    and potentially your MPI job)
        slurmstepd: *** STEP 4560.0 ON mexico48 CANCELLED AT 2017-10-25T19:28:20 ***
        srun: Job step aborted: Waiting up to 32 seconds for job step to finish.
        srun: error: mexico48: tasks 0-1,3-4: Killed
        srun: error: mexico48: task 2: Exited with exit code 5
        srun: error: mexico49: tasks 5-9: Killed
```
On top of that, the report file is not created, and the trace file is created but empty. On both compute nodes.

- To be "sure" the problem does not stem from the version of **openmpi** I am using, I tried to compile the tutorial without involving **GEOPM**, and I ran the tutorial. Everything works just fine.

Any idea ?

Thanks in advance,
--Mathieu.

cmcantalupo, 2017-10-25:
Thank you for the detailed feedback!  The issue of why the launcher failed I know:  we parse the node list option and our current parser assumes a comma separated list (it doesn't understand the bracket notation), so "-w mexico48,mexico49" will work but "-w mexico[48,49]" does not.  We will try to fix this issue soon.  

The invalid communicator error I'm not so sure what is going wrong.  Have you configured geopm with the --enable-debug option?  If not, can you recompile with this option and report back if any error messages re emitted?  

Thanks,
Chris

matsKNY, 2017-10-26:
Thank you for your anwser Christopher.

To begin with, I tried to specify the nodes without the bracket notation, and the output remains the same, namely `<geopmpy.launcher> list index out of range`.
I have tried to just specify the nodes to the launcher, and it seems that the argument processing is correct:
```bash
user@mexico48 $ geopmsrun -w mexico[48,49]
        srun -w mexico[48,49]
        srun: fatal: No command given to execute.

user@mexico48 $ geopmsrun -w mexico48,mexico49
        srun -w mexico48,mexico49
        srun: fatal: No command given to execute.
```
However, the nodelist is not completely "enforced". In fact, I checked if Slurm job had been executed and it appears the `lscpu` command has been executed on **mexico51**, before the error message is printed:
```bash
user@mexico48 $ scontrol show job
        JobId=4569 JobName=lscpu
           UserId=root(0) GroupId=root(0)
           Priority=4294901666 Nice=0 Account=root QOS=normal
           JobState=COMPLETED Reason=None Dependency=(null)
           Requeue=1 Restarts=0 BatchFlag=0 Reboot=0 ExitCode=0:0
           RunTime=00:00:00 TimeLimit=UNLIMITED TimeMin=N/A
           SubmitTime=2017-10-26T11:20:59 EligibleTime=2017-10-26T11:20:59
           StartTime=2017-10-26T11:20:59 EndTime=2017-10-26T11:20:59
           PreemptTime=None SuspendTime=None SecsPreSuspend=0
           Partition=All AllocNode:Sid=mexico48:13104
           ReqNodeList=(null) ExcNodeList=(null)
           NodeList=mexico51
           BatchHost=mexico51
```

Then, I recompiled **GEOPM** and the tutorials with the `--enable-debug` option, but I get the same output as before, no additional error message or trace. Is there a way to check the option **GEOPM** was compiled with, just to be sure the `--enable-debug` option has been taken into account ? I "parsed" the output of the compilation and the `-O0 -g` flags are used, so it seems to be compiled with the debug  option activated.

Thank you in advance,
--Mathieu.

bgeltz, 2017-10-28:
Hi Mathieu,

I've started digging into this a bit, and was wondering if you could answer the following for me:

What version of SLURM are you using currently?
What version of GEOPM are you using currently?  The output from "git describe" would be great.
Is POSIX shared memory enabled on your system?

There's a patch Chris has up for review now that you could cherrypick to remove the Fortran tests.  That should avoid the compilation issue you were hitting.  That's here:
https://review.gerrithub.io/#/c/384000/

I was just testing geopmsrun on our cluster, and it appears to be honoring the -w parameter for me.  I'm trying to see if I can reproduce what you were hitting.  I'll post back when I get some more information.  I'm not seeing the "list index out of range" error yet.

The issue with lscpu that you saw is that the particular section of code that issues the command does not enforce the nodelist you've specified.  There's a patch Chris just wrote that's up for review that should address that.  It's here:
https://review.gerrithub.io/#/c/384140/

The peculiar MPI error that was shown when you tried to use MPIEXEC directly is curious.  I'm wondering if there's something misconfigured with MPI that's resulting in that error.  I think it would be illuminating if you tried to run our unit tests and post the output back here.  They will expose any MPI issues that may be present in the system when they run.  To do that simply run "make check" from the root of your checkout.

Good luck,
Brad

cmcantalupo, 2017-10-28:
A simple way to determine if debug is enabled is the following command:

    $ grep enable_debug config.log
    enable_debug='1'

enable_debug should be set to '1'.  That being said, if the code is compiling with -O0 -g then debug is enabled.  I had forgotten that in order to get a stack trace from the python script with errors in python you must also

    $ export GEOPM_DEBUG=true

If you do this python errors will emit the stack trace were the "list index out of range" error occurred in the python tools.  I'd be very interested to see where this is happening.  

With the patch to pass the node list to the lscpu call, there is a warning emitted by slurm because we only launch lscpu on one node, but the node list given may have more than one node.  Warnings aside, with the patch to add the node list to the lscpu command seems to properly restrict the node use.

The issue with the brace notation verses the comma separated list is actually not an problem when the underlying job launcher is slurm so the 

    -w mexico[48,49] 

option should actually work fine.  We only try to parse this option as a comma separated list for the Intel MPI launcher.

Hope that helps,
Chris

matsKNY, 2017-11-02:
Thank both of you for your answers.

I will try to be as thorough as possible:

- I confirm that the debug option of GEOPM is enabled.

- The version of Slurm currently installed on our development cluster is **15.08.12**:
```bash
user@mexico48 $ sinfo --version
    slurm 15.08.12
```

- The version of GEOPM being currently used is **0.3.0**. I cherry-picked the two commits Brad mentioned before executing the command. I can undo it and redo the `git describe`, if needed:
```bash
user@mexico48 $ git describe
    v0.3.0-88-g9d77cb2
```

- I think POSIX shared memory is enabled since a `tmpfs` is mounted on `/dev/shm`. On top of that, 0% of the allocated 32 GB is used. Another way to check ?
```bash
user@mexico48 $ mount | grep shm
    tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev)
```

- I tried to re-run the launching script for the tutorial 0 for `srun`, just adding `-w mexico48,mexico49`. To be thorough, I tried running it while mexico[48,49] were reserved, and while they were idle (I am speaking of Slurm node states). Both time, the same error message:
```bash
user@mexico48 $ export GEOPM_DEBUG=true
user@mexico48 $ ./tutorial_0.sh
    srun: Warning: can't run 1 processes on 2 nodes, setting nnodes to 1
    srun: error: Unable to allocate resources: Requested nodes not in this partition
    Traceback (most recent call last):
      File "/scratch/stoffelm/geopm/build/bin/geopmsrun", line 38, in <module>
        err = geopmpy.launcher.main()
      File "/scratch/stoffelm/geopm/build/lib/python2.7/site-packages/geopmpy/launcher.py", line 1230, in main
        launcher = factory(sys.argv)
      File "/scratch/stoffelm/geopm/build/lib/python2.7/site-packages/geopmpy/launcher.py", line 115, in factory
        time_limit, job_name, node_list, host_file)
      File "/scratch/stoffelm/geopm/build/lib/python2.7/site-packages/geopmpy/launcher.py", line 776, in __init__
        time_limit, job_name, node_list, host_file)
      File "/scratch/stoffelm/geopm/build/lib/python2.7/site-packages/geopmpy/launcher.py", line 462, in __init__
        self.init_topo()
      File "/scratch/stoffelm/geopm/build/lib/python2.7/site-packages/geopmpy/launcher.py", line 568, in init_topo
        launcher.run(stdout=ostream)
      File "/scratch/stoffelm/geopm/build/lib/python2.7/site-packages/geopmpy/launcher.py", line 533, in run
        raise subprocess.CalledProcessError(pid.returncode, argv_mod)
    subprocess.CalledProcessError: Command 'srun -N 1 -n 1 -w mexico48,mexico49 lscpu' returned non-zero exit status 1
```
As Chris said, there is a warning message. The error message mentions partitions, that is why I tried to run the script while the nodes were in different states according to Slurm. The nodes are in the same partition, named `All`, which is up. 

- I tried to re-compile the code after cherry-picking the purge of the Fortran tests, no problem at all. Thank you for the fix.

- I executed the `make check` command, and 6 tests out of the 176 failed, producing the same error message (content of the logs related to the failed tests in `gtests_link`, and printed on the terminal):
```bash
Failed tests:
    FAIL: test/gtest_links/MPITreeCommunicatorTest.hello
    FAIL: test/gtest_links/MPITreeCommunicatorTest.send_policy_down
    FAIL: test/gtest_links/MPITreeCommunicatorTest.send_sample_up
    FAIL: test/gtest_links/MPIProfileTest.noctl
    FAIL: test/gtest_links/MPIInterfaceTest.geopm_api
    FAIL: test/gtest_links/MPIInterfaceTest.mpi_api

Error message:
    --------------------------------------------------------------------------
    mpiexec has detected an attempt to run as root.
    Running at root is *strongly* discouraged as any mistake (e.g., in
    defining TMPDIR) or bug can result in catastrophic damage to the OS
    file system, leaving your system in an unusable state.

    You can override this protection by adding the --allow-run-as-root
    option to your cmd line. However, we reiterate our strong advice
    against doing so - please do so at your own risk.
    --------------------------------------------------------------------------
```
Is there an easy way to specify the `--allow-run-as-root` option to all the tests, in order to see if there is another problem concerning MPI ? Or another workaround ?

Thank you in advance for your answers,
--Mathieu.

bgeltz, 2017-11-27:
My apologies for the tardiness in replying here.  SC kept everyone very busy this month.  Replies below:

> The version of Slurm currently installed on our development cluster is 15.08.12:
> 
> user@mexico48 $ sinfo --version
>     slurm 15.08.12

We've been testing internally with v15.08.7 so this should be fine.

> The version of GEOPM being currently used is 0.3.0. I cherry-picked the two commits Brad mentioned before executing the command. I can undo it and redo the git describe, if needed:
>
> user@mexico48 $ git describe
>     v0.3.0-88-g9d77cb2

Your version looks good.  Those commits you needed have since been merged into the dev branch which is available here: [dev branch](/geopm/geopm/tree/dev)

> I think POSIX shared memory is enabled since a tmpfs is mounted on /dev/shm.

This appears correct to me.

> I tried to re-run the launching script for the tutorial 0 for srun, just adding -w mexico48,mexico49. To be thorough, I tried running it while mexico[48,49] were reserved, and while they were idle (I am speaking of Slurm node states). Both time, the same error message:
>
> user@mexico48 $ export GEOPM_DEBUG=true
> user@mexico48 $ ./tutorial_0.sh
>     srun: Warning: can't run 1 processes on 2 nodes, setting nnodes to 1
>     srun: error: Unable to allocate resources: Requested nodes not in this partition
...
> As Chris said, there is a warning message. The error message mentions partitions, that is why I tried to run the script while the nodes were in different states according to Slurm. The nodes are in the same partition, named All, which is up.

I believe the problem here is that you ask for specific nodes with -w, but while those nodes are in the "All" partition all is not the default partition.  Have you tried using the -p option with -w in this case?  Like:

    srun -p <PARTITION_NAME> -w <NODE_LIST> ...

> I executed the make check command, and 6 tests out of the 176 failed

The failures you have listed are all expected if MPI is not working properly.  While you could modify the launcher.py to include the allow-run-as-root flag, this could be dangerous.  If the cluster you're operating on is yours, and you're not worried about screwing things up as root which would require you to re-image the system / start from scratch then this is not a big concern.  If on the other hand you share this system with many other people and you're not willing to risk screwing up the machine for a few hours or days, then I'd recommend against running MPI as root.

Instead I'd start down the patch of installing the msr-safe kernel driver on your cluster.  The whole problem here is that in order to read/write from the MSR device files, you need to be root.  The msr-safe project allows privileged users read and write access to a whitelist of MSRs and bitfields within those MSRs for very fine grained control.  We've been using it on all the systems where GEOPM research is performed and it is our preferred setup.  Here's that project:

[msr-safe](/LLNL/msr-safe)

If you need assistance setting up msr-safe let me know.  There's a script to build an RPM of the driver here:

https://github.com/LLNL/msr-safe/blob/master/rpm/build_rpm.sh

You should be able to use that to build the RPM on the system in question.  After that, installing the RPM should setup the `msr` user, move the kernel object out to the right spot, and setup the initial whitelist.  When the driver is loaded, the `/dev/cpu/*/msr_safe` device files should be available which GEOPM looks for.

I hope this helps!  Let me know if you hit anymore roadblocks.

Brad

matsKNY, 2017-12-04:
Hello Brad,

First of all, thank you for your answer. I hope everything went well at SC17.

Concerning the launching through `geompsrun`, it was a stupid mistake, which is on me. I had a reservation on the nodes I tried to launch the job onto, thus they were not available (since the reservation is not yet taken into account by `geopmsrun`).
However, before I realized that, I tried what you suggested. It failed with the following output:
``` bash
<geopmpy.launcher> Command 'srun -N 1 -n 1 -w mexico48,mexico49 lscpu' returned non-zero exit status 1
```
Since there is no mention of the `-p All` part, I am not sure whether it is taken into account by `geopmsrun` or not.

Concerning the "msr-safe" part, I am still facing roadblocks. In a few key points:

- I cloned the `msr-safe` repository, compiled and installed all the components on the compute nodes, following the `README` embedded in the concerned repository;
- I created the group `msr` and added my user account and the `slurm` account to this group;
- I checked that the generated whitelist corresponds to the CPU model embedded in the compute nodes (Haswell - 0x063f);
- I wrote a really simple (almost dumb) code snippet to verify that I can access the MSR through the `msr-safe` driver, from user mode ([the snippet](https://gist.github.com/anonymous/e51d0f6b35664f149f8ff189fb6682ab)). You can compile it with `gcc -o test_msr-safe test_msr-safe.c` and run it with `./test_msr-safe`. It will simply read and print the content of the TSC register. Nonetheless, the file `/dev/cpu/0/msr-safe` is opened with the `O_RDWR` mode, as it is done in `GEOPM/PlatformImp.cpp`; 

- I tried to run the first tutorial as I had previously done:
``` bash
user@mexico48 $ export GEOPM_RM=NONE
user@mexico48 $ export MPIEXEC="srun --reservation=Power-GEOPM --exclusive --cpu_bind=v,mask_cpu:0x1,0x2,0x4,0x8,0x10 -N 2 -n 10"
user@mexico48 $ ./tutorial_0.sh
```
This time, `GEOPM` seems not to be able to open the `/dev/cpu/0/msr-safe` file due to a permission issue:
``` bash
cpu_bind=MASK - mexico49, task  8  3 [7540]: mask 0x8 set
cpu_bind=MASK - mexico49, task  9  4 [7541]: mask 0x10 set
cpu_bind=MASK - mexico49, task  7  2 [7539]: mask 0x4 set
cpu_bind=MASK - mexico49, task  5  0 [7537]: mask 0x1 set
cpu_bind=MASK - mexico49, task  6  1 [7538]: mask 0x2 set
cpu_bind=MASK - mexico48, task  1  1 [45210]: mask 0x2 set
cpu_bind=MASK - mexico48, task  2  2 [45211]: mask 0x4 set
cpu_bind=MASK - mexico48, task  0  0 [45209]: mask 0x1 set
cpu_bind=MASK - mexico48, task  4  4 [45213]: mask 0x10 set
cpu_bind=MASK - mexico48, task  3  3 [45212]: mask 0x8 set

Error: <geopm> Could not open MSR device: permission denied opening device /dev/cpu/0/msr_safe: at geopm/src/PlatformImp.cpp:376
```
- To rule out a potential `slurm`-related issue, I tried to launch the code snippet mentioned above thanks to `slurm`:
```bash
user@mexico48 $ srun --reservation=Power-GEOPM -w mexico48 ./test_msr-safe

About the MSR:
    Address: 0x10
    Content: 0x3b49ae435d1966

```
It seems to be properly functioning. I also checked that the permissions, groups and owners associated with the executables (`tutorial_0` and `test_msr-safe`) are identical.

Have you any idea of the root of the issue?

Thank you in advance for your answers.

slabasan, 2017-12-15:
Hi Mathieu, 

My name is Stephanie, I'm one of the msr-safe code maintainers here at LLNL. I was reading through your comments here and it looks like msr-safe is the culprit here. I see that your simple code snippet works on mexico48. As another sanity check, when you run your code snippet on mexico49, are you still able to access the TSC register? One hunch here is that msr-safe was installed on one node, but will also need to be installed on all nodes where you want to access MSRs. Generally, we see this error message when permissions do not allow access to the `msr_safe` device files.

bgeltz, 2017-12-18:
> Since there is no mention of the -p All part, I am not sure whether it is taken into account by geopmsrun or not.

You're right, presently we don't take -p into account properly in the launcher.  There's a pre-run step that will try to issue _lscpu_ on the compute nodes prior to actually running your job.  This is done to determine the proper affinity mask for your system's topology.  Chris has a patch that's he's testing to address this and it's here:

https://review.gerrithub.io/#/c/392225/

If you cherrypick that it should address issues around handling -p.  Let me know if you're having issues.

The other thing you might want to check on is the permissions for the msr_safe file descriptor.  Like Stephanie said above, you'll want to make sure you have msr_safe installed in all the nodes where you want to run.  You can run a command like the one below to see if the msr_safe file is there and what the permissions are on it.  It should be owned by root, and have group r/w access for the msr group.

```
srun -N1 -p <PARTITION> -w <TEST_NODE> ls -l /dev/cpu/*/msr_safe
```

matsKNY, 2017-12-20:
Hello Stephanie and Brad,

Thank you for your answers. You were right, the problem I encountered stemed from the installation of the `msr-safe` driver on mexico49. I followed the same installation procedure on both nodes, and tested the code snippet only on mexico48 (kind of a stupid mistake). On mexico49, I only checked with `systemctl` that the daemon associated with `msr-safe` was running.
It seems that the multipath associated with the underlaying disk arrays failed during the installation on mexico49, resulting in msr-safe rpm not being able to set properly the access rights/permissions/ownerships on the  `msr-safe` driver files.

However, I am now facing a shared memory-related issue:
```bash
Error: No such file or directory: SharedMemoryUser: Could not open shared memory with key "/geopm-shm-16130-comm-split-prof": at geopm/src/SharedMemory.cpp:166
```
 I checked on each compute node, the shared memory is active and seems to be properly functioning (`geopm` is able to create `geopm-shm-16130-sample`, `geopm-shm-16130-tprof`, and `openmpi-sessions-16130@mexico49_0` in `/dev/shm`).

After looking to the source code of `geopm`, I would say the issue is related to the timeout timer set up for the instance of the `ShareMemoryUser` class. However, the timeout seems to be set up to 30 seconds by default, and the error message usually show up after a couple of seconds (I timed it). On top of that, the tutorial 0, out of the tens of executions I tried, successfully executed 4 times. Have you got any idea concerning this issue ?

Finally, concerning `geopmsrun`, the patch Christopher coded solves the issue concerning the preliminary `lscpu` job. However, I am now facing this error message:
```bash
<geopmpy.launcher> sequence item 8: expected string, NoneType found
```
Hereinbelow, the way I modified the `tutorial_0.sh` launching script:
```bash
geopmsrun  -p All \                                                         
           -w mexico48,mexico49 \                                           
           -N 2 \                                                           
           -n 8 \                                                           
           --geopm-preload \                                                
           --geopm-ctl=process \                                            
           --geopm-report=tutorial_0_report \                               
           --geopm-trace=tutorial_0_trace \                                 
           -- ./tutorial_0
```
Thank you in advance for your answers,
--Mathieu.

bgeltz, 2018-01-24:
My apologies for the tardiness in replying on this one.  We've had a particularly busy start to the year.

> However, I am now facing a shared memory-related issue:
> 
> Error: No such file or directory: SharedMemoryUser: Could not open shared memory with key "/geopm-shm-16130-comm-split-prof": at geopm/src/SharedMemory.cpp:166

I'm starting to see this now testing on our internal cluster with the GNU toolchain.  It's very sporadic for me.  It seems to flip flop between consistently failing and consistently passing with no clear indication as to why that's happening.  I'm only seeing this when I use the GNU toolchain.

I've never see it when I use the Intel toolchain.  Our standard environment is to use the Intel toolchain, MVAPICH2, and autotools.  You might have more success in the short term if you try that.

I'm currently digging into the failure with GNU in the loop and will update this when I know more.

Usages of `geopmsrun` should be fixed as of the last release v0.4.0.  Please take a look at the latest tip of the dev branch (973044a) and let me know if you're still experiencing issues.

bgeltz, 2018-01-31:
I just put a patch up to workaround the code where this problem has been occurring.  If you're still bumping into this issue on your cluster than this may solve the problem:

https://review.gerrithub.io/#/c/397299/

This does require an MPI implementation that supports MPI_Comm_split_type to function properly.  If you're using OpenMPI you need to use at least v1.7.4 where this API is implemented.  I am currently testing with MVAPICH2 2.2 and it seems to resolve the issue so far.  

Prior to this patch the error about not being able to open the shared memory key is exceptionally sporadic using the GNU toolchain and MVAPICH2.  I will report back after further testing.

closed 2018-02-08 by cmcantalupo

bgeltz, 2018-02-09:
@matsKNY As far as I can tell, we've resolved the issues we were having with shared memory.  The problem ended up being a combination of issues in our code (for the application launch method of the Controller specifically), and on at least one system an issue with the way systemd was configured on the compute nodes.

systemd has a configuration file located at `/etc/systemd/logind.conf` which may have an option set called `RemoveIPC`.  If `RemoveIPC=yes` is set as either a compile time default or an override in that file, then systemd will remove shared memory keys created by GEOPM which will cause the issue you describe above.  You need to verify that `RemoveIPC=no` is set on the compute nodes for things to work properly.  I've updated the README on the GitHub front page to reflect these findings which is available [here](https://github.com/geopm/geopm#systemd-configuration).

If you have any other problems, please feel free to open a new issue or reach out via email:
brad.geltz (at) intel.com.

commit 9e402dad (2018-02-08): Fix shared memory issues during startup.

- The application launch method was causing issues if the Controller's
  process came up in the middle of the app's processes.  In that case,
  the Controller would remove all shared memory keys in it's
  constructor.  This caused errors in initializing the app's shared
  memory if one of the app's ranks had already created it's shared
  memory objects.
- Try to unlink all stale keys before creating new ones.
- Remove erroneous double split of the Controller's comm.  The ppn1 comm
  is already passed into the constructor.
- Add unit test to expose shme [... cut]