geopm/geopm #112: Improper shutdown if MPI application has stray popen

maiterth, 2017-10-26:
Hi,
I encountered the following problem:
Running geopmsrun with an application results in the 'proper' execution of the MPI appliation, but geopm, never returns. 
The trace file is not populated and the report file is not generated.
 
I tracked down the the problem to  all call of the Signal exception: (./configure --enable-debug helped.)
"Error: File exists: Signal 17 raised." ( thrown by SignalException::SignalException() )

I found that the application does properly handle signals, but apparently one of the linked libraries of the MPI application doesn't . 
The assumption is popen() is called while not properly handling SIGCHLD in the library. But GEOPM is waiting for the last process to be terminated since it does handle signals properly.

A workaround would be to ignore SIGCHLD in the signal handler SignalHandler::SignalHandler()
But this comes with its disadvantages.

Thanks,

cmcantalupo, 2017-10-28:
Proposal to stop handling SIGCHLD is up here:

https://review.gerrithub.io/#/c/383994/

Please feel free to comment.

bgeltz, 2017-11-14:
The patch to address this had been merged.  Closing.

closed 2017-11-14 by bgeltz