geopm/geopm #113: ompt-overlapping-error

bgeltz, 2017-11-15:
On Quartz, when using `--enable-ompt` when I configure, I see this when I try to launch jobs:

```
Error: <geopm> Logic error: Error parsing /proc/self/maps, overlapping address ranges.: at geopm/src/OMPT.cpp:143
```

It's happening 100% of the time at the moment.  It's preventing data from being written to the report files at the end of execution.

bgeltz, 2018-03-01:
I just saw this again when trying to run tutorial_2.sh having built it and GEOPM with the GNU toolchain:

```
$ ./tutorial_2.sh 
Warning: User requested all cores for application. GEOPM controller will share a core with the OS.

GEOPM_TRACE=tutorial_2_trace GEOPM_PROFILE= LD_LIBRARY_PATH=/usr/workspace/wsb/variorum/intelCRADA/geopm/BUILD/lib/geopm/openmp/lib:/usr/workspace/wsb/variorum/intelCRADA/geopm/BUILD/lib:/p/lscratchh/geltz2/build/geopm/lib:/usr/workspace/wsb/variorum/intelCRADA/geopm/BUILD/lib:/usr/tce/packages/mkl/mkl-2018.0/lib:/usr/tce/packages/mvapich2/mvapich2-2.2-intel-16.0.3/lib:/usr/tce/packages/intel/intel-16.0.3/lib/intel64:/p/lscratchh/geltz2/build/geopm/lib:/p/lscratchh/geltz2/build/geopm/lib LD_DYNAMIC_WEAK=true GEOPM_REPORT=tutorial_2_report OMP_PROC_BIND=true GEOPM_PMPI_CTL=process MV2_ENABLE_AFFINITY=0 KMP_AFFINITY=disabled srun -N 2 -n 10 --mpibind=off --cpu_bind v,mask_cpu:0x1,0x1ff0000001ff,0x3fe0000003fe00,0x7fc0000007fc0000,0xff8000000ff8000000 ./tutorial_2

cpu_bind=MASK - quartz17, task  7  2 [140169]: mask 0x3fe0000003fe00 set
cpu_bind=MASK - quartz17, task  8  3 [140170]: mask 0x7fc0000007fc0000 set
cpu_bind=MASK - quartz17, task  9  4 [140171]: mask 0xff8000000ff8000000 set
cpu_bind=MASK - quartz17, task  6  1 [140168]: mask 0x1ff0000001ff set
cpu_bind=MASK - quartz17, task  5  0 [140167]: mask 0x1 set
cpu_bind=MASK - quartz16, task  2  2 [107644]: mask 0x3fe0000003fe00 set
cpu_bind=MASK - quartz16, task  3  3 [107645]: mask 0x7fc0000007fc0000 set
cpu_bind=MASK - quartz16, task  4  4 [107646]: mask 0xff8000000ff8000000 set
cpu_bind=MASK - quartz16, task  1  1 [107643]: mask 0x1ff0000001ff set
cpu_bind=MASK - quartz16, task  0  0 [107642]: mask 0x1 set
MPI_COMM_WORLD size: 8
Beginning loop of 10 iterations.
OMP: Warning #181: OMP_PROC_BIND: ignored because KMP_AFFINITY has been defined
OMP: Warning #181: OMP_PROC_BIND: ignored because KMP_AFFINITY has been defined
OMP: Warning #181: OMP_PROC_BIND: ignored because KMP_AFFINITY has been defined
OMP: Warning #181: OMP_PROC_BIND: ignored because KMP_AFFINITY has been defined
OMP: Warning #181: OMP_PROC_BIND: ignored because KMP_AFFINITY has been defined
OMP: Warning #181: OMP_PROC_BIND: ignored because KMP_AFFINITY has been defined
OMP: Warning #181: OMP_PROC_BIND: ignored because KMP_AFFINITY has been defined
OMP: Warning #181: OMP_PROC_BIND: ignored because KMP_AFFINITY has been defined
Completed loop.                    
Error: <geopm> Logic error: Error parsing /proc/self/maps, overlapping address ranges.: at geopm/src/OMPT.cpp:130
Error: <geopm> Logic error: Error parsing /proc/self/maps, overlapping address ranges.: at geopm/src/OMPT.cpp:130
srun: error: quartz16: task 0: Exited with exit code 254
srun: error: quartz17: task 5: Exited with exit code 254
<geopmpy.launcher> Command 'srun -N 2 -n 10 --mpibind=off --cpu_bind v,mask_cpu:0x1,0x1ff0000001ff,0x3fe0000003fe00,0x7fc0000007fc0000,0xff8000000ff8000000 ./tutorial_2' returned non-zero exit status 254```

amarathe84, 2019-03-06:
After a bit of debugging on Quartz around this issue, I found that /usr/lib64/ld-2.17.so and [vdso] address ranges as seen in /proc/self/maps share a byte address (end of ld-2.17.so and start of [vdso]). Here's a screen dump of some of the variables in OMPT constructor that I printed out: 

```
addr_begin:2aaaaaaab000 addr_end:2aaaaaacd000   object:/usr/lib64/ld-2.17.so
addr_begin:2aaaaaacd000 addr_end:2aaaaaacf000   object:[vdso]
``` 

On Quartz I've been using a workaround in src/OMPT.cpp where I skip throwing the exception if the entry being parsed is "[vdso]". It fixes the issue for now since this is the only (potentially harmless) case of address overlap on Quartz. But it's obviously not a generic solution, so I'm looking for one.

cmcantalupo, 2019-07-10:
Looks like libelf is LGPL and may provide the features we need.

cmcantalupo, 2019-08-20:
The license ambiguity is resolved by the RedHat distribution of elfutils:

https://android.googlesource.com/platform/external/elfutils/+/refs/heads/jb-mr1.1-dev-plus-aosp/EXCEPTION

This seems to explicitly say that you can link to the API's declared in libelf.h under any open source license.  This is not distributed as part of the upstream source here:

https://sourceware.org/git/?p=elfutils.git;a=tree

The documentation for libelf is not awesome.  Looks like the best there is is this pdf:

https://sourceforge.net/projects/elftoolchain/files/Documentation/libelf-by-example/

cmcantalupo, 2019-08-20:
I don't think libelf is the right tool.  Looks like::

dladdr(3)

is all we need, something like:

```
#define _GNU_SOURCE
#include <dlfcn.h>

std::string address_to_name(void *addr)
{
    Dl_info info;
    int err = dladdr(addr, &info);
    std::string result(info.dli_fname);
    result += ":";
    result += info.dli_sname;
    return result;
}

```

cmcantalupo, 2019-08-20:
starting to test:

https://github.com/cmcantalupo/geopm/commit/403efc15fe2445f58522a1ff8bc660f3c68b2076

cmcantalupo, 2019-08-21:
https://review.gerrithub.io/c/geopm/geopm/+/465834

commit 5f001f40 (2019-09-17): Use libdl and libelf to map instruction address to symbol name

- Replacing implementation that required popen() to be called
  at runtime.
- This is not allowed for some MPI varients (e.g. Cray MPI).
- New implemenation calls libdl for dynamic symbols.
- For non-dynamic symbols libelf is used to resolve the address.
- Helper routines are provided that return symbol from address,
  and offset symbol map from path.
- Fixes #113

Change-Id: Idfaf0f5a3f79cac00f4c39fc4f96aa092e89024c
closed 2019-09-17 by cmcantalupo