geopm/geopm #129: Add support for more platforms in PlatformIO

cmcantalupo, 2017-12-11:
The MSR array defined at the bottom of PlatformIO.cpp is only filled out for the KNL platform.  Add support for other architectures like BDX and HSX.

cmcantalupo, 2017-12-11:
@slabasan I think you mentioned that you had done this for some platforms other than KNL.  If so can you share the code?  Thanks!

slabasan, 2017-12-11:
I started to populate the SNB MSR list on the `slabasan-populate-snb-msrs` branch here: [https://github.com/slabasan/geopm/tree/slabasan-populate-snb-msrs].

slabasan, 2017-12-22:
@cmcantalupo I had some updates to my `slabasan-populate-snb-msrs` branch, but I am not sure how to update the existing change on Gerrit. Is it as simple as doing a `git push gerrit slabasan-populate-snb-msrs`, where `gerrit` is `https://review.gerrithub.io/geopm/geopm`?

dianarg, 2018-09-27:
SNB, BDX/HSX are supported in PlatformIO.  Support for SKX has also been added.  Updates from @slabasan regarding the PERF_STATUS domain for Haswell were incorporated at one point (later changed from CPU to CORE).  Thanks Stephanie!

https://review.gerrithub.io/c/geopm/geopm/+/391682
https://review.gerrithub.io/c/geopm/geopm/+/398837
https://review.gerrithub.io/c/geopm/geopm/+/411487

closed 2018-09-27 by dianarg