geopm/geopm #134: Support python 3.x

cmcantalupo, 2017-12-14:
We are currently only supporting python 2.7.  Upgrade code so that it is python 3.x compatible.

dianarg, 2018-07-13:
We also need to add a setting in .travis.yml so that it runs the tests with both 2.7 and 3.

dianarg, 2018-10-31:
Ideally this will make all python code version-agnostic.

cmcantalupo, 2018-12-11:
https://github.com/openhpc/ohpc/pull/877

dannosliwcd, 2019-10-16:
We have been gradually working toward 2+3 compatibility over the past few months. Since 6ac9fc45a, automation is in place to run unit tests on 2.7 and 3.6, and to run a python 3 compatibility linter, though full support for 3.x is not yet complete.

**Next steps**:
- Try to use python 3 where possible to identify anything that may have been missed
- Identify a packaging and release plan. Do we want to release for just one version, or both? Consider that a typical user is more likely to gain access through the full project install (e.g. in an rpm), rather than through pip.
- Remove 2.7-only references:
  - specfile install paths, PYTHONPATH recommendations in documentation (depending on the packaging plan)
  - python_requires, and pip categorization tags in setup.py
- To combat gotchas from changing the default python from 2 to 3, RHEL8 packaging prevents python shebangs that are missing an explicit version number. We will need to avoid errors like `*** ERROR: ambiguous python shebang in /usr/bin/geopmconvertreport: #!/usr/bin/python. Change it to python3 (or python2) explicitly.` If dual 2+3 support is desired, it may be worthwhile to release separate geopmpy packages that explicitly use python 2 or python 3.

dannosliwcd, 2019-10-25:
Use 3.6 as the default. Let users manually install for 2.7 if they want

commit 83483b3f (2019-11-05): Modify specfiles to explicitly use either python 2 or python 3

- Resolves #134

Change-Id: I5f1c44852b4301ee8b8aa5a51714e7c3f31deb65
closed 2019-11-05 by cmcantalupo