geopm/geopm #181: Update Plugin Developer's Guide to reflect new interfaces

dianarg, 2018-03-01:
The guide should describe the new interfaces used to write plugins using PlatformIO and updates to how plugins work. 

Other important notes that should be mentioned in the updated guide:
- Naming conventions for plugins, signal and control names, and signal/control aliases.
- The difference between region runtime and telemetry timestamp.
- Relationship between Deciders and PlatformIO
- List of built-in high level signal/control aliases (this should also be in man pages)
- How sampling works and what determines frequency of samples

dianarg, 2018-05-16:
For beta, documentation will be covered by new tutorials for the IOGroup and Agent.

bgeltz, 2018-05-24:
We have tutorials for this now.  Also, the old guide has been removed.

closed 2018-05-24 by bgeltz