geopm/geopm #192: Support for overriding settings from environment with values from disk

dianarg, 2018-03-08:


cmcantalupo, 2019-03-21:
I'm not sure why this issue was closed, but I'd like to reopen it.   The following would be useful in the context of enabling system administrator control of some parameters for geopm:

Two optional configuration files can be added.  Suggested location:

/var/lib/geopm/environment-override.json
/var/lib/geopm/environment-default.json

An example file would be contain the following text:
{
    "GEOPM_AGENT": "power_balancer"
    "GEOPM_POLICY": "/var/lib/geopm/policy.json"
}

If this was the environment-override.json, every contronroller run on that compute node would execute the power balancer agent with the policy file specified.  This could be used to enforce a power cap on the system.

tpatki, 2019-04-17:
As per our discussion, here are my thoughts and what I think we should commit to for the first cut. 
[SLURM.docx](https://github.com/geopm/geopm/files/3091371/SLURM.docx)

tpatki, 2019-04-17:
Yes. This is what we're intending to commit for June, mostly capturing discussions with Chris and adding clarity so we don't (1) start with too many broad assumptions, and (2) overcommit and not be able to deliver the milestone on time.

commit 06344573 (2019-05-07): Update Environment class to initialize from json on disk

- Fixes #192
- Update unit tests

Change-Id: I75bebe3bf274be5efcfafc58cee948a0773684c7
closed 2019-05-07 by cmcantalupo