geopm/geopm #202: registering plugins in IOGroup based on underlying architecture

milpuz, 2018-03-15:
In register_built_in_once() in IOGroup class the plugins are registered regardless of underlying architecture. This means that when registering plugin MSRIOGroup under Power microarchitecture we fail to register it because MSR registers do not exists on that architecture, but instead of gracefully failing (i.e. reporting error and continuing without reporting metrics as defined by MSRIOGroup or running MPI application without GEOPM or shutting down application with error) application just hangs. Should the architecture on which GEOPM is running be detected and based on it only plugins that are appropriate loaded? For now in Power port we use ifdefs to disable loading of plugins that are not needed/required.

The same comments goes when registering platforms in PlatformFactory constructor, but I guess this code will go once the features from Platform/PlatformImp classes are ported to PlatformIO?

dianarg, 2018-06-14:
Currently, a warning is printed if the MSRIOGroup could not be loaded.  To fix this correctly, we need a list of signals and controls required by PlatformIO.  Then the warning would only be printed if those required signals/controls were not provided by some IOGroup.

Note that REGION_ID# is one of these required signals, but it can only be provided once we have connected to an application.  We will need to have special handing for the region id signal.

cmcantalupo, 2019-04-16:
I'm removing the bug label, and promoting to a feature.  Basics of bug were fixed with:
a5e5ec91bdb11c9