geopm/geopm #207: alterantive to popen() when running MPI with OpenFabrics support

milpuz, 2018-03-15:
When running application under GEOPM on OpenMPI 3 that uses OpenFabrics/OpenIB software stack the following working is printed out:

```
--------------------------------------------------------------------------
A process has executed an operation involving a call to the
"fork()" system call to create a child process.  Open MPI is currently
operating in a condition that could result in memory corruption or
other system errors; your job may hang, crash, or produce silent
data corruption.  The use of fork() (or system() or other calls that
create child processes) is strongly discouraged.

The process that invoked fork was:

  Local host:          [[18735,1],0] (PID 68176)

If you are *absolutely sure* that your application will successfully
and correctly survive a call to fork(), you may disable this warning
by setting the mpi_warn_on_fork MCA parameter to 0.
--------------------------------------------------------------------------
```
This warning due to partial/lack of support for sytem call within the OpenFabrics/OpenIB stack as explained [here](http://www.open-mpi.org/faq/?category=openfabrics#ofa-fork).

It would be useful to obtain information received from 'lscpu -x' in PlatformTopo.cpp in some other way instead of using popen() to execute it? Otherwise this warning can be supressed by doing mpirun with `--mca mpi_warn_on_fork 0`.

drvadim, 2018-06-11:
Hi, We are getting the following error on one of the Power8 boxes:
```
Error: <geopm> Runtime error: PlatformTopo: parsing lscpu output, key not found: "CPU(s)": at geopm/src/PlatformTopo.cpp:285
Error: <geopm> Runtime error
```
It happens on one box out of four, and we cannot figure out what's the trigger. Any pointers for further debugging are appreciated.

cmcantalupo, 2018-07-23:
I have posted a patch up for review *may* solve this issue:

https://review.gerrithub.io/c/geopm/geopm/+/420201

Please let me know if this approach helps.

cmcantalupo, 2018-07-24:
The patch here:

https://review.gerrithub.io/c/geopm/geopm/+/420201

was merged into the dev branch.  I will close this issue.  @drvadim: please let me know if the problem persists.  Thanks.

closed 2018-07-24 by cmcantalupo