geopm/geopm #208: parsing lscpu output in presence of SMT

milpuz, 2018-03-15:
There is validation check at https://github.com/geopm/geopm/blob/21b435f15b1810c865f3d234397b72a1ceab55e4/src/PlatformTopo.cpp#L227 that multiples number of sockets with number of cores per socket with threads per core and compares the value with number of CPUs obtained from lscpu output. When SMT is disabled on PowerPC the output from `lscpu -x` is:

```
Architecture:          ppc64le
Byte Order:            Little Endian
CPU(s):                160
On-line CPU(s) mask:   0x101010101010101010101010101010101010101
Off-line CPU(s) mask:  0xfefefefefefefefefefefefefefefefefefefefe
Thread(s) per core:    1
Core(s) per socket:    10
Socket(s):             2
NUMA node(s):          2
Model:                 1.0 (pvr 004c 0100)
Model name:            POWER8NVL (raw), altivec supported
CPU max MHz:           4023.0000
CPU min MHz:           2394.0000
Hypervisor vendor:     (null)
Virtualization type:   full
L1d cache:             64K
L1i cache:             32K
L2 cache:              512K
L3 cache:              8192K
NUMA node0 CPU(s):     0x1010101010101010101
NUMA node1 CPU(s):     0x101010101010101010100000000000000000000
```

In this case above validation would fail because terms do not multiply to 160 but to 20. To fix this issue the on-line CPU mask should be used to count number of CPUs instead of CPU(s) number.

milpuz, 2018-03-20:
The below commit counts number of online CPUs and uses it to test when reading from CPU(s) fails:

https://github.com/milpuz/geopm/commit/5b0cb38a6c47e821a001ad45f800473a82b4d2c0#diff-d76a1eb2df16d63ba60388107aad4b3a

bgeltz, 2018-04-24:
The loop here:
https://github.com/milpuz/geopm/commit/5b0cb38a6c47e821a001ad45f800473a82b4d2c0#diff-d76a1eb2df16d63ba60388107aad4b3aR241

Should be implemented more like this one:
https://github.com/geopm/geopm/blob/dev/src/PlatformTopo.cpp#L321

I'll be cooking up a patch for this shortly.

bgeltz, 2018-04-25:
Patch is up for review here:
https://review.gerrithub.io/#/c/geopm/geopm/+/409080/

commit e0e59988 (2018-04-28): Fix lscpu parsing for offline CPUs.

- Fixes #208.
- Based on a patch by Milos:
  + milpuz/geopm@5b0cb38#diff-d76a1eb2df16d63ba60388107aad4b3a

Change-Id: Ie454a90e9ac934e7793fc920182e3e4933db08b5
closed 2018-04-28 by cmcantalupo