geopm/geopm #225: Add environment variable and necessary logic to handle the sample output to the resource manager.

bgeltz, 2018-04-07:
This should compliment geopm_env_policy().

bgeltz, 2018-05-01:
This task should change slightly.  @cmcantalupo is thinking something like the following:

1. Add GEOPM_ENDPOINT env var handling that will serve as part of the base name for the policy and sample files.

2. Policy file will be something like: /geopm-shm-${GEOPM_ENDPOINT}.policy

3. Sample file will be something like /geopm-shm-${GEOPM_ENDPOINT}.sample

If GEOPM_POLICY is set to a JSON file, use that.  GEOPM_POLICY should not be able to create shmem regions.  Should not be able to read a policy from JSON and write samples to shmem.

It's either read the policy and write the samples to shmem, or read the policy from JSON and do not provide the sample data to anyone.

bgeltz, 2018-10-31:
There are 2 things that appear conflated here:  reading and writing the policy from wither JSON or shmem, and having default agent parameters available on request.

dianarg, 2019-10-18:
Reopening to address the following: 
- GEOPM_POLICY will be used for file policies
- GEOPM_ENDPOINT will be used for the endpoint shmem prefix that provides a policy
- GEOPM_ENDPOINT takes precendence *unless* GEOPM_ENDPOINT is in the environment default only but the user environment provided GEOPM_POLICY.  Normal rules for environment override still apply.

This patch covers this change: https://review.gerrithub.io/c/geopm/geopm/+/471413

dianarg, 2019-11-20:
WW45: Under review here: https://review.gerrithub.io/c/geopm/geopm/+/471413

commit f7a9ec8b (2019-12-06): Separate use of GEOPM_POLICY for file policy only

- GEOPM_ENDPOINT should be used for the shmem endpoint name.
- GEOPM_POLICY is for file policies only.
- If both are provided, GEOPM_ENDPOINT takes precendence.
- EndpointUser timeout for #728 will be addressed separately.
- Resolves #225

Change-Id: I44ac3706eb5040385fbb58a095241d425399ba8e
closed 2019-12-06 by cmcantalupo