geopm/geopm #244: Incorrect frequency (percent of sticker) values in report

dianarg, 2018-05-09:
The frequency values in the report are based on the CYCLES_REFERENCE and CYCLES_THREAD signals.  Currently, the report shows very large positive or negative values, but these percentages should all be in a range from 0 to ~120%.

bgeltz, 2018-06-14:
One of the problems with the patch that was previously merged to address this is that msr-safe currently prevents the zeroing out of any of the fixed counters based on the default whitelist.  None of the fixed counters are writable, so these lines do not work:

https://github.com/geopm/geopm/blob/dev/src/MSRIOGroup.cpp#L674-L676

There are a few other issues I've uncovered with respect to MSR overflow and fixed counter usage that I've addressed in a patch that's currently up for review:

https://review.gerrithub.io/c/geopm/geopm/+/413871

This patch does not completely solve the issue of erroneous values showing up in the reports w.r.t. frequency data.  I still see negative values and impossibly large values intermittently.  I am actively debugging this issue.  YMMV presently if you need to study the frequency achieved for a given region based on the data that is in the report files.

Until this issue is completely solved, there may be intermittent issues in frequency reporting.

cmcantalupo, 2018-06-16:
This issue is resolved by the commits below:

https://github.com/geopm/geopm/commit/0b967ae353462b90be84078eac772beccd116686
https://github.com/geopm/geopm/commit/69954a6edb136a5f30b86364ac15da677ee11768
https://github.com/geopm/geopm/commit/b1531e0528244517c476cf859c6ae0eb1fc72424

commit d0a5338b (2018-05-23): Initialize fixed counters at MSRIOGroup ctor time.

- Zero out fixed counters at start up.
- Print frequency in report in Hz.
- Print runtime as per-rank-average as well as
  collective/synchronized runtime.
- Fixes #244
- Fixes #245

Change-Id: I6a82f67118563cdf72ff3a1b3bd699d29435b3d5
closed 2018-06-16 by cmcantalupo