geopm/geopm #249: Prove package scoped MSR based controls are working

bgeltz, 2018-05-15:
The geopmread/write testing is exposing a potential issue where writing a control to a package scoped MSR may not actually result in the desired change at the CPU level.

This may result in a change to all our package scoped controls such that they write to all CPUs instead of just one.

Controlling frequency via IA32_PERF_CTL and IA32_PERF_STATUS were used to uncover this.

dianarg, 2018-05-17:
Confirmed that setting frequencies for individual CPUs works on mcfly (SKX).  On quartz, I am not seeing the frequencies I expect. 

On KNL, setting package frequencies works with this patch: https://review.gerrithub.io/#/c/geopm/geopm/+/411494/

dianarg, 2018-05-21:
Package scoped controls fixed by: c2e6a9199f89e4984c74d2640e7821551f518b13
CPU controls changed to core scope by: 1e659437f8665364df7dc76c64da8aac3b900afd

closed 2018-05-21 by dianarg