geopm/geopm #255: Add whitelist generation using geopmread/geopmwrite

dianarg, 2018-05-29:
This replaces the feature provided by geopmpolicy.

bgeltz, 2018-10-31:
We need a new tool for this if we're actually going to do this.  It doesn't make sense to have this wrapped into geopmread/write because it's specific to msr-safe.  It isn't generic to all IOGroups.

This usefulness is questionable.

For diagnostic purposes, an integration test could be written to read all the MSRs that GEOPM could read and verify that msr-safe allowed that to complete OK.  Similarly for all the controls, sane values could be determined programmatically to test that msr-safe allows the write to complete.

dianarg, 2020-09-27:
This feature is now available from the `geopmadmin` tool.

closed 2020-09-27 by dianarg