geopm/geopm #260: GEOPM_DEBUG_ATTACH broken for non-numeric values

bgeltz, 2018-06-07:
When attempting to utilize GEOPM_DEBUG_ATTACH with a non-numeric value, the Environment parser throws:

terminate called after throwing an instance of 'geopm::Exception'
  what():  <geopm> Invalid argument: Environment::Environment(): Value could not be converted to an integer: at geopm/src/Environment.cpp:218

Afterward, the controller exits without every coming up fully.  The documentation here:

https://github.com/geopm/geopm/blob/dev/ronn/geopm.7.ronn

Suggests that: If set, but not to a numerical value then all ranks are trapped.  The functionality appears broken.

bgeltz, 2018-06-14:
```
geopmsrun -N 1 -n 4 --geopm-debug-attach --geopm-ctl=process --geopm-report=t6_report --geopm-trace=t6_trace -- ./tutorial_6 tutorial_6_config.json
<geopmpy.launcher> --geopm-ctl must be one of: "process", "pthread", or "application"
```

That should work.  If I pass a number it will trap that rank.  If I pass something like true, it still barfs:

```
geopmsrun -N 1 -n 4 --geopm-debug-attach true --geopm-ctl=process --geopm-repo
rt=t6_report --geopm-trace=t6_trace -- ./tutorial_6 tutorial_6_config.json
...
terminate called after throwing an instance of 'geopm::Exception'
  what():  <geopm> Invalid argument: Environment::Environment(): Value could not be converted to an integer: at geopm/src/Environment.cpp:218
...
```

sidjana, 2020-03-16:
TODO: Also, change the error message to indicate the name of the env. variable

dianarg, 2020-04-01:
Change implementation to match documentation.

PR #1167 (2020-06-19): Fix --geopm-debug-attach command line option to match the documentation
- Fixes #260 from github issues
commit 380649f3 (2020-06-19): Fix --geopm-debug-attach command line option to match the documentation

- Fixes #260 from github issues

Change-Id: I19192de37f449388cfed0893b9d7f0144ddff5d8
closed 2020-06-19 by bgeltz