geopm/geopm #271: Report region data is incorrect for energy counters

bgeltz, 2018-06-19:
I'm seeing wildly varying data for energy in the new reports.  The energy reporting for regions appears to be an entire order of magnitude smaller utilizing the new code path.

This command was used to generate data with the new code path:
```
$ GEOPM_RM="SLURM" GEOPM_AGENT="energy_efficient" GEOPM_KEEP_FILES=true GEOPM_RUN_LONG_TESTS=true ./geopm_test_integration.py -v TestIntegration.test_plugin_efficient_freq_offline
```

The resulting report data for dgemm at sticker looks like the following with the new code path:
```
##### geopm 0.5.0+dev4gb1531e0 #####
Profile: test_plugin_efficient_freq_offline_freq_2200000000.0
Agent: energy_efficient
Policy Mode: deprecated
Tree Decider: deprecated
Leaf Decider: deprecated
Power Budget: -1

Host: quartz17
Final freq map:
...
Region dgemm (0x00000000a74bbf35):
    runtime (sec): 9.76876
    sync-runtime (sec): 0.980965
    energy (joules): 71.6641
    frequency (%): 117.014
    frequency (Hz): 2.45729e+09
    mpi-runtime (sec): 0
    count: 10
```

The same data utilizing the old code path (by dropping GEOPM_AGENT= from the above command):

```
##### geopm 0.5.0+dev4gb1531e0 #####
Profile: test_plugin_efficient_freq_offline_freq_2200000000.0
Policy Mode: DYNAMIC
Tree Decider: static_policy
Leaf Decider: efficient_freq
Power Budget: 400

Host: quartz17
...
Region dgemm (11396693813):
    runtime (sec): 10.1416
    energy (joules): 2660.76
    frequency (%): 111.705
    mpi-runtime (sec): 0.1402
    count: 10
```

So the fact that we see 71.6 Joules in the new code path and 2660.7 in the old code path is a problem given that the runtimes/frequencies are comparable.  

To top it off, the energy data is highly variable in the new code path:

```
------------------------------------------------------------------------------------------------------------------------
Region : dgemm
                   energy   runtime  mpi_runtime  frequency  count      power  energy_savings  runtime_savings
region freq_mhz                                                                                               
dgemm  2200       71.6641   9.76876          0.0   117.0140     10   7.336049      -54.449718       -71.234630
       2100      214.2650  10.85000          0.0    99.9999     10  19.747926       48.342006       -54.170507
       2000      103.2270  10.65970          0.0    95.2385     10   9.683856       -7.224854       -56.922803
       1900      112.2010  11.79040          0.0    90.4752     10   9.516301        1.351147       -41.873897
       1800      105.3340  11.94850          0.0    85.7144     10   8.815667       -5.080031       -39.996652
       1700      112.1910  12.94230          0.0    80.9507     10   8.668552        1.342354       -29.246734
       1600      197.7510  12.94480          0.0    76.1904     10  15.276482       44.028096       -29.221772
       1500      204.9880  13.69540          0.0    71.4287     10  14.967653       46.004156       -22.139551
       1400       86.0456  15.06910          0.0    66.6653     10   5.710069      -28.635282       -11.005302
       1300      119.2880  17.05070          0.0    61.9047     10   6.996076        7.211958         1.895523
       1200      136.9350  18.13110          0.0    57.1429     10   7.552493       19.169679         7.741395

```

That data is not as variable, and seems valid in the old code path:

```
------------------------------------------------------------------------------------------------------------------------
Region : dgemm
                  energy  runtime  mpi_runtime  frequency  count       power  energy_savings  runtime_savings
region freq_mhz                                                                                              
dgemm  2200      2660.76  10.1416     0.140200   111.7050     10  262.360969       -0.640419        -2.487773
       2100      2708.71  10.2680     0.209060    99.9998     10  263.801130        1.141134        -1.226139
       2000      2712.47  11.4023     0.204586    95.2378     10  237.887970        1.278171         8.843830
       1900      2726.94  11.9181     0.207437    90.4760     10  228.806605        1.802020        12.788951
       1800      2702.34  11.9875     0.163365    85.7141     10  225.429823        0.908102        13.293848
       1700      2682.88  12.1119     0.173344    80.9524     10  221.507773        0.189349        14.184397
       1600      2763.56  13.2495     0.210122    76.1905     10  208.578437        3.103244        21.552511
       1500      2837.74  14.3384     0.203234    71.4285     10  197.911901        5.636175        27.510043
       1400      2831.55  14.6065     0.245878    66.6667     10  193.855475        5.429888        28.840585
       1300      2841.29  15.7289     0.216183    61.9048     10  180.641367        5.754076        33.918456
       1200      2871.28  17.6608     0.210809    57.1429     10  162.579272        6.738458        41.147060

```

bgeltz, 2018-06-19:
I've been running the test_counters test like this:
```
$ git clean -fd . ; GEOPM_RM="SLURM" GEOPM_AGENT="energy_efficient" GEOPM_KEEP_FILES=true GEOPM_RUN_LONG_TESTS=true ./geopm_test_integration.py -v TestIntegration.test_counters
```

You should not have to set ```GEOPM_AGENT``` on the cmdline since it's being set in the Python code explicitly.

bgeltz, 2018-06-19:
Setting the following in the Python has no discernible effect on the report data:

```
os.environ['GEOPM_REGION_BARRIER'] = 'true'
```

dianarg, 2018-06-20:
On BDX, I added the energy for the two sockets and two DRAMs as separate region signals in the report.  I see a lot of energy for specific regions being attributed to the unmarked region.

Data for tutorial 2 with all2all removed (report is below):
```
region    board energy    total pkg + dram energy
stream    2749.76         3660.66
dgemm     317.91          736.2936
sleep     129.767         286.5424
unmarked  1631.96	  145.9054
-------------------------------------------
          4829.397        4829.4014

epoch     4935.93	
application  4938.25	
```

Unfortunately, it looks like #268 is going to be as easy to fix as I originally thought.  The signal index passed to push_region_signal_total() might already be a combined signal, and we don't have access to the original domain of that index.  It might easier to do the separation into packages and dram in the Reporter itself for now.

```
Host: quartz2391
Region tutorial_stream (0x000000008a1e10cd):
    runtime (sec): 48.4436
    sync-runtime (sec): 35.195
    energy (joules): 2749.76
    pkg0 energy: 1059.71
    pkg1 energy: 1027.32
    dram0 energy: 779.724
    dram1 energy: 793.906
    frequency (%): 121.623
    frequency (Hz): 2.55409e+09
    mpi-runtime (sec): 0
    count: 20
Region tutorial_dgemm (0x00000000c82f28bd):
    runtime (sec): 11.4615
    sync-runtime (sec): 4.81618
    energy (joules): 317.91
    pkg0 energy: 267.554
    pkg1 energy: 275.831
    dram0 energy: 91.5246
    dram1 energy: 101.384
    frequency (%): 123.515
    frequency (Hz): 2.59382e+09
    mpi-runtime (sec): 0
    count: 10
Region tutorial_sleep (0x000000009803a79a):
    runtime (sec): 10.0007
    sync-runtime (sec): 4.57106
    energy (joules): 129.767
    pkg0 energy: 121.738
    pkg1 energy: 67.8481
    dram0 energy: 72.8128
    dram1 energy: 24.1435
    frequency (%): 133.738
    frequency (Hz): 2.80849e+09
    mpi-runtime (sec): 0
    count: 10
Region unmarked-region (0x2000000000000000):
    runtime (sec): 0.830744
    sync-runtime (sec): 26.1063
    energy (joules): 1631.96
    pkg0 energy: 45.8457
    pkg1 energy: 43.3051
    dram0 energy: 29.0767
    dram1 energy: 27.6779
    frequency (%): 123.712
    frequency (Hz): 2.59796e+09
    mpi-runtime (sec): 0
    count: 0
Region epoch (0x8000000000000000):
    runtime (sec): 70.6882
    sync-runtime (sec): 0
    energy (joules): 4935.93
    pkg0 energy: 0
    pkg1 energy: 0
    dram0 energy: 0
    dram1 energy: 0
    frequency (%): 0
    frequency (Hz): 0
    mpi-runtime (sec): 0
    count: 10
Application Totals:
    runtime (sec): 70.7488
    energy (joules): 4938.25
    mpi-runtime (sec): 0
    ignore-time (sec): 0
    geopmctl memory HWM: 48740 kB
    geopmctl network BW (B/sec): 0
```

dianarg, 2018-06-20:
Another observation that may be related: dram_power calculated by test_power_consumption is much higher with the new code path.  This means that the PowerGovernorAgent is setting the package power too low.

Old code path:
```
Power stats from mr-fusion4 :
       pkg_energy-0  dram_energy-0  elapsed_time  socket_power    dram_power  combined_power
count  34499.000000   34499.000000  34499.000000  34499.000000  34499.000000    34499.000000
mean       0.629812       0.092118      0.005039    125.012259     18.279863      143.292122
std        0.035661       0.011998      0.000209      5.008140      2.184784        6.013367
min        0.424255       0.016006      0.004251     84.175843      3.167558       89.360195
25%        0.621216       0.086975      0.004981    123.582553     17.308418      141.860013
50%        0.628479       0.092773      0.005020    125.321168     18.470292      143.743081
75%        0.636780       0.097885      0.005057    127.090035     19.502354      145.615627
max        3.555176       0.630341      0.029355    168.229446     27.134881      187.082938
```

New code path:
```
Power stats from mr-fusion4 :
       pkg_energy-0  dram_energy-0  elapsed_time  socket_power    dram_power  combined_power
count  11120.000000   11120.000000  11120.000000  11120.000000  11120.000000    11120.000000
mean       5.114180       2.049280      0.051632    100.212922     40.724075      140.936998
std        2.248777       0.796504      0.024497      5.653610      5.411243       10.074201
min        0.621521       0.136963      0.007284     62.798389     16.875578       88.106139
25%        3.906937       1.615112      0.038542     98.175190     38.721556      137.606252
50%        4.425476       1.824554      0.044191    100.293620     41.074132      141.714073
75%        5.444366       2.207748      0.055695    102.007892     43.076482      144.752890
max       25.912903       8.842773      0.282382    204.670129    110.097180      313.564808
```

cmcantalupo, 2018-06-21:
If the above numbers are measured on KNL, is it possible that on-package-memory is being used in one case and not in the other?  The usual way of directing the application to use on-package-memory is with "numactl -m1" preceding the application call.

bgeltz, 2018-06-25:
This is fixed with v0.5.1, https://github.com/geopm/geopm/tree/v0.5.1.

closed 2018-06-25 by bgeltz