geopm/geopm #308: Integration tests pick parameters based on head node cpuid

cmcantalupo, 2018-07-19:
On Theta where the head nodes are Xeon and the compute nodes are XeonPhi the integration tests choose power limits for Xeon.  

https://github.com/geopm/geopm/blob/dev/test_integration/geopm_test_integration.py#L65-L72

cmcantalupo, 2018-07-20:
Hacky fix is up here:

https://github.com/cmcantalupo/geopm/commit/18b93b80a53b0f0c9afa381ac505612e47c1be3f

hopefully the hash sticks around.

bgeltz, 2018-11-16:
Not sure if there's a better solution than what Chris suggested above.  That solution checks the hostname to see if we're running on Theta, and makes the necessary changes.

I could write a decorator (or perhaps the use the setUp() method) that tries to aprun and inspect ```/proc/cpuinfo``` to get the CPU family/model information and from there set it for the test.  It would only be valid for ALPS, so I would still have to have some sort of detection logic for that in the test_integration.py so it would know to aprun the inspection code.

Is that any better or worse than just checking for "theta" in the hostname and hardcoding the model/family info?

cmcantalupo, 2018-11-16:
Check how lscpu is executed on the compute node by geopmpy/launcher.py,

commit c14b479c (2018-11-30): Integration test decorators implemented with launcher invocation.

- This allows for the checks to be ran on a compute node rather
  than head/login/mom nodes.
- Fixes #308.

Change-Id: Iebaa2f67c31668bd6b13bad1a54c2b42b3baa71e
closed 2018-11-30 by cmcantalupo