geopm/geopm #314: Prepare shared space on Theta for easy GEOPM integration with COBALT scripts

bgeltz, 2018-07-26:


cmcantalupo, 2018-08-09:
/projects/intel/geopm-home/build/geopm-dev.sh
/projects/intel/geopm-home/build/geopm-master.sh

closed 2018-08-09 by cmcantalupo

cmcantalupo, 2018-08-29:
A simple way to integrate with geopm on Theta is to add:

    source /projects/intel/geopm-home/build/geopm-dev.sh

To your cobalt script.  This will add the GEOPM installed paths to your environment.  If you want to look through the documentation then source the script as above in your bash shell and type:

    man geopm
    man geopmaprun

cmcantalupo, 2018-08-30:
Attached is a special version of the tutorials designed specifically for running on ALCF's Theta including COBALT scripts:

[geopm-tutorial.tar.gz](https://github.com/geopm/geopm/files/2334303/geopm-tutorial.tar.gz)

Also note that this uses the new Agent code path.

cmcantalupo, 2018-08-30:
The modified geopm tutorial in tarball form can be found on Theta here:
 
/projects/intel/geopm-home/geopm-tutorial.tar.gz
 
I have compiled and submitted a few jobs and the output from the compile and run can be found on Theta here:
 
/projects/intel/geopm-home/geopm-tutorial
 
Users should be able to:
 
    cp /projects/intel/geopm-home/geopm-tutorial.tar.gz .
    tar xvf geopm-tutorial.tar.gz
    cd geopm-tutorial/
    ./tutorial_build_theta.sh
    qsub tutorial_0.cobalt
 
this will build the tutorials and submit tutorial 0 to the cobalt queue.  Note that users will need to change the account in the cobalt script from “Intel” to whatever account they want to charge the job to.