geopm/geopm #316: Add policy columns to the power_balancer trace output

cmcantalupo, 2018-07-27:
Add columns to the trace that give the values of the policy that was last passed down.

bakerbrandond, 2018-07-28:
I'll do this. I'll add it as a patch dependent on my current refactor endeavors. Would rather not have to track a branch or worry about merge conflicts for something so easy

cmcantalupo, 2018-07-28:
Okay.  I was thinking one column per policy value.

policy_power_cap
policy_step_count
policy_max_epoch_runtime
policy_power_slack

Also the current power limit enforced at the leaf perhaps as the last column:

policy_power_limit

Seems like having a generic solution to add the Agent policy values to the trace might be useful.  I'll look into that briefly.

commit 77f96969 (2018-09-07): Augment trace columns of the PowerBalancerAgent.

- Add last received policy values to the trace.
- Add enforced power limit to the trace.
- Fixes #316.

Change-Id: I52994b0167a1bb3d5cbbdc31577012a928fe6bf8
closed 2018-09-07 by cmcantalupo