geopm/geopm #320: Select performance governor with srun in launcher

cmcantalupo, 2018-08-01:
The srun command can accept an option --cpu-freq=Performance to run the job under the performance governor.  This option should be set by the geopmpy Launcher.  We currently have a requirement in our README that the performance governor always be enabled on the system.  This change would relax that requirement on slurm systems that support this option Linux power management to be enabled on compute nodes which are not running GEOPM.   

Thanks to Joe Crawford for making this suggestion.

bgeltz, 2018-08-06:
FWIW, there's also a ```CpuFreqDef``` we could specify in the slurm.conf:
>CpuFreqDef
Default CPU frequency governor to use when running a job step if it has not been explicitly set with the --cpu-freq option. Acceptable values at present include:
Conservative
attempts to use the Conservative CPU governor
OnDemand
attempts to use the OnDemand CPU governor
Performance
attempts to use the Performance CPU governor
PowerSave
attempts to use the PowerSave CPU governor

https://slurm.schedmd.com/slurm.conf.html

cmcantalupo, 2018-08-09:
https://review.gerrithub.io/c/geopm/geopm/+/421030

closed 2018-08-13 by bgeltz