geopm/geopm #329: libhwloc deps not available on XC40

fearful-symmetry, 2018-08-10:
We're running into some issues trying to get geopm installed on our compute nodes:
```
zypper in geopm-0.5.1-1.1.x86_64.rpm
Loading repository data...
Reading installed packages...
Resolving package dependencies...

Problem: nothing provides libhwloc.so.0()(64bit) needed by geopm-0.5.1-1.1.x86_64
 Solution 1: do not install geopm-0.5.1-1.1.x86_64
 Solution 2: break geopm-0.5.1-1.1.x86_64 by ignoring some of its dependencies

Choose from above solutions by number or cancel [1/2/c] (c): 
```

The only libhwloc package available on Cray-SLES: 
```
zypper search libhwloc
Loading repository data...
Reading installed packages...

S | Name      | Summary                     | Type   
--+-----------+-----------------------------+--------
i | libhwloc5 | Runtime libraries for hwloc | package
```

this provides `libhwloc.so.5`, which isn't symlinked to `libhwloc.so.0`, so any zypper/rpm install will fail.

I tried manually symlinking the two, which _seems_ to work, but that's a temporary solution at best, Cray's image building tools won't be happy with it. Anyone else run into this? 
I can't find any rpm that provides `libhwloc.so.0` on Fedora 28 or RHEL.

cmcantalupo, 2018-08-10:
Did the RPM come from here?

http://download.opensuse.org/repositories/home:/cmcantalupo:/geopm/SLE_12_SP3/x86_64/

If so, this RPM was built off of the version of hwloc I packaged for SLES:

https://download.opensuse.org/repositories/home:/cmcantalupo/SLE_12_SP3/x86_64/

Seems like SLES has a more up to date version of hwloc that they support. 

    S  | Name        | Summary                                            | Type   
    ---+-------------+----------------------------------------------------+--------
    i+ | hwloc       | Portable Hardware Locality                         | package
    i  | hwloc-data  | Run time data for hwloc                            | package
    i+ | hwloc-devel | Headers and shared development libraries for hwloc | package
       | hwloc-doc   | Documentation for hwloc                            | package
    i+ | libhwloc5   | Runtime libraries for hwloc                        | package

That is the output I get from the Theta Cray system at ANL when I zypper search.  The BuidRequires field in the geopm spec file requests hwloc-devel which does exist on the Cray SLES:

https://github.com/geopm/geopm/blob/dev/geopm.spec.in#L44

Can you run the make rpm target from the geopm Makefile?  I can try to remove my package from OBS and see if it picks up the default SLES version.

cmcantalupo, 2018-08-10:
I have confirmed that if I remove my packaging of hwloc from my OBS instance OBS is unable to find hwloc-devel in the default OBS SLES-SP3 repo.  I'll look into getting an up to date SLES package into my OBS as a dependency.

cmcantalupo, 2018-08-10:
If the download fails the Makefile will touch an empty file for the gmock archive.  If you run

    rm gmock-1.7.0.zip geopm-*.tar.gz
    make dist
    make rpm

does that help?

fearful-symmetry, 2018-08-13:
This is ongoing. Having issues with failed tests no, so `make rpm` is dying:

```
FAIL: test/gtest_links/ManagerIOTest.negative_write_json_file
FAIL: test/gtest_links/ManagerIOSamplerTest.negative_bad_files
[  FAILED  ] ManagerIOSamplerTest.negative_bad_files (1 ms)
[  FAILED  ] 1 test, listed below:
[  FAILED  ] ManagerIOSamplerTest.negative_bad_files
 1 FAILED TEST
[  FAILED  ] ManagerIOTest.negative_write_json_file (0 ms)
[  FAILED  ] 1 test, listed below:
[  FAILED  ] ManagerIOTest.negative_write_json_file
 1 FAILED TEST
```

bgeltz, 2018-08-13:
Do you happen to have the remainder of the log for the above failures?  I'm hoping there's a little more information regarding the errors when these tests failed.  I can't tell why they failed just yet.

I've been using the gist system for sharing snippets like that:
https://gist.github.com/

cmcantalupo, 2018-08-13:
So clearing out the distribution tarball seemed to fix the issue around the empty gmock archive embedded in the distribution tarball.  Glad to hear you got around that issue.  

I'm not entirely sure why the unit tests are failing in the rpmbuild.  One option is to disable the unit tests during building the rpm:

https://github.com/cmcantalupo/geopm/tree/cmcantalupo-rpmbuild-no-check

Not sure that is the best solution, but it will get you past this issue, and running the unit tests during the RPM build is not entirely standard.  If you can print out the log from that failure we can dig into the failure.  The detailed error message should be lower in the log after the failure.

fearful-symmetry, 2018-08-13:
The individual logs from the failed tests aren't too interesting:
```
Note: Google Test filter = ManagerIOTest.negative_write_json_file
[==========] Running 1 test from 1 test case.
[----------] Global test environment set-up.
[----------] 1 test from ManagerIOTest
[ RUN      ] ManagerIOTest.negative_write_json_file
test/ManagerIOTest.cpp:150: Failure
Expected to throw, but succeeded.
[  FAILED  ] ManagerIOTest.negative_write_json_file (6 ms)
[----------] 1 test from ManagerIOTest (6 ms total)

[----------] Global test environment tear-down
[==========] 1 test from 1 test case ran. (7 ms total)
[  PASSED  ] 0 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] ManagerIOTest.negative_write_json_file

 1 FAILED TEST

<?xml version="1.0" encoding="UTF-8"?>
<testsuites tests="1" failures="1" disabled="0" errors="0" timestamp="2018-08-13T15:51:52" time="0.007" name="AllTests">
  <testsuite name="ManagerIOTest" tests="1" failures="1" disabled="0" errors="0" time="0.006">
    <testcase name="negative_write_json_file" status="run" time="0.006" classname="ManagerIOTest">
      <failure message="test/ManagerIOTest.cpp:150&#x0A;Expected to throw, but succeeded." type=""><![CDATA[test/ManagerIOTest.cpp:150
Expected to throw, but succeeded.]]></failure>
    </testcase>
  </testsuite>
</testsuites>

```
```
Note: Google Test filter = ManagerIOSamplerTest.negative_bad_files
[==========] Running 1 test from 1 test case.
[----------] Global test environment set-up.
[----------] 1 test from ManagerIOSamplerTest
[ RUN      ] ManagerIOSamplerTest.negative_bad_files
test/ManagerIOTest.cpp:349: Failure
Exception message was different from expected: <geopm> Invalid argument: ManagerIOSampler::read_file(): input configuration file invalid: at geopm/src/ManagerIO.cpp:261
Expected message: input configuration file "ManagerIOSamplerTest_empty" could not be opened
[  FAILED  ] ManagerIOSamplerTest.negative_bad_files (9 ms)
[----------] 1 test from ManagerIOSamplerTest (9 ms total)

[----------] Global test environment tear-down
[==========] 1 test from 1 test case ran. (11 ms total)
[  PASSED  ] 0 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] ManagerIOSamplerTest.negative_bad_files

 1 FAILED TEST
<?xml version="1.0" encoding="UTF-8"?>
<testsuites tests="1" failures="1" disabled="0" errors="0" timestamp="2018-08-13T15:51:53" time="0.011" name="AllTests">
  <testsuite name="ManagerIOSamplerTest" tests="1" failures="1" disabled="0" errors="0" time="0.009">
    <testcase name="negative_bad_files" status="run" time="0.009" classname="ManagerIOSamplerTest">
      <failure message="test/ManagerIOTest.cpp:349&#x0A;Exception message was different from expected: &lt;geopm&gt; Invalid argument: ManagerIOSampler::read_file(): input configuration file invalid: at geopm/src/ManagerIO.cpp:261&#x0A;Expected message: input configuration file &quot;ManagerIOSamplerTest_empty&quot; could not be opened" type=""><![CDATA[test/ManagerIOTest.cpp:349
Exception message was different from expected: <geopm> Invalid argument: ManagerIOSampler::read_file(): input configuration file invalid: at geopm/src/ManagerIO.cpp:261
Expected message: input configuration file "ManagerIOSamplerTest_empty" could not be opened]]></failure>
    </testcase>
  </testsuite>
</testsuites>
```
I was about to do what @cmcantalupo suggested and just disable tests in the Makefile.

fearful-symmetry, 2018-08-13:
After disabling tests, `make rpm` works just fine. As expected, it produces a dependency with what's available on Cray SLES
```
rpm -qpR /usr/src/packages/RPMS/x86_64/geopm-0.5.1+dev31g09e482d-1.x86_64.rpm | grep hwloc
libhwloc.so.5()(64bit)
```
It installs fine via zypper.

bgeltz, 2018-08-13:
When you were trying the ```make rpm``` previously and seeing the unit test failures, were you trying to do that as root?

The tests failed in a peculiar way that make it seem to me that they were being executed as the root user.

fearful-symmetry, 2018-08-13:
ha, yah...that works. Our computes are accessed via a root shell, so I didn't bother su'ing to myself.

bgeltz, 2018-08-13:
Ah, ok.  Running the tests as root is something I've not tried, so it's good to know there's something we need to document and fix in that space.  Glad to see things are resolved.

fearful-symmetry, 2018-08-14:
Any updates on the packaging situation?

cmcantalupo, 2018-08-14:
## COPIED FROM SLACK

I'm glad you managed to get a package building.  Before I go down the path of trying to publish RPMs in the public OBS instance that are useful for your application, I think I'd like to know better what that application is .  In particular, are you trying to build RPMs to install on a shared compute system like Theta that uses a module system to provide different compiler toolchains and MPI implementations?  If so we would need to be building GEOPM RPMs that support the combinations of toolchain and MPI implementation selected by the module.  I'm less concerned about the compiler toolchain than I am about the MPI implementation selection.  By default the GEOPM spec file builds GEOPM against the default SLES install of OpenMPI.  On theta, however, the default MPI selection is cray-mpich/7.7.2.  I think we want to compile against the Cray MPICH install and not the SLES OpenMPI.  You can see that we have published RPMs for all of the compiler tool / MPI implementation combinations for OpenHPC:

http://build.openhpc.community/OpenHPC:/1.3:/Update6:/Factory/SLE_12/x86_64/

This uses the whole OpenHPC infrastructure, and I probably not exactly what you want in the context of a Cray standard install.

I can grab a source RPM for hwloc from a recent SLES system and try to publish it on my OBS, but let me know if that is really sufficient with regard to the MPI dependency issues.

## END OF COPY FROM SLACK.

In going further down the rabbit hole of Cray packaging I have found the following:

    $ rpm -qf /opt/cray/pe/mpt/7.7.2/gni/mpich-cray/8.6/lib/libmpich.so
    file /opt/cray/pe/mpt/7.7.2/gni/mpich-cray/8.6/lib/libmpich.so is not owned by any package

So it seems that the mpich library that is exposed by

    module load mpich-cray

is not known to the RPM database on the head node of Theta.  Not sure how to set up the proper BuildRequires so that we can create a spec file appropriate for a Cray install if the files we depend on are not known to the RPM database.

fearful-symmetry, 2018-08-15:
Yah, this is all for theta. I'm honestly not sure how to handle this, but it sounds like we might have to do different builds for the different MPI environments.

On your second comment:

you're right, the `mpich-cray` libs won't be on the RPM database. Cray creates the environment by overlaying a squashfs image built from a chroot onto the head node.

How do we want to proceed as far as builds?

cmcantalupo, 2018-08-15:
Looking through the module files available on Theta, it seems that swapping the PrgEnv module will change the compiler tool chain, but all PrgEnv modules seem to point to the cray-mpich MPI implementation.  I think we can support this one option for MPI pretty easily.  I am going to start to put together a spec file targeting Theta based on our openhpc spec file that generates a lmod module file:

https://github.com/geopm/geopm/blob/dev/geopm-ohpc.spec.in

I have opened a new issue to track this change:

https://github.com/geopm/geopm/issues/335

and will post the branch I'm working on there to get feedback.

fearful-symmetry, 2018-08-15:
Ah, that's good to kknow.

cmcantalupo, 2018-08-15:
Going to close this issue and track this problem here:

https://github.com/geopm/geopm/issues/335

moving forward.

cmcantalupo, 2018-08-16:
Going to reopen this until the SLES packages on OBS install cleanly.  We will be dropping the hwloc dependency entirely when this patch:

https://review.gerrithub.io/c/geopm/geopm/+/409755

is merged.  Then I can just remove the hwloc packaging from OBS.

bakerbrandond, 2018-10-29:
Closed by 716c72e2

closed 2018-10-29 by bakerbrandond