geopm/geopm #335: Create spec file for Theta installation

cmcantalupo, 2018-08-15:
Modify OpenHPC spec file to support Cray Linux Environment (CLE) targeting the Theta machine at ALCF.

fearful-symmetry, 2018-08-15:
So, full disclosure, I'm pretty bad with specfiles, but it seems fine. 

The only thing that pops out at me is the install dir, perhaps create a macro like
`%define softdir soft` for the sake of being more agnostic?

cmcantalupo, 2018-08-15:
The script below builds an RPM on theta that installs to the directories appropriate for Theta including a module file for GEOPM.  The output of the script can be found in my home directory on Theta:

    /home/cantalup/rpmbuild/RPMS

I have put the changes up for review here:

https://review.gerrithub.io/c/geopm/geopm/+/422463

Comments or questions are welcome.

```
#!/bin/bash
echo WARNING: THIS SCRIPT DELETES \"~/rpmbuild\" AND
echo          RUNS \"git clean -dfx\" IN THE geopm-rpm REPO.

sleep 3

set -e
set -x

if [ ! -d geopm-rpm ]; then
    git clone https://github.com/cmcantalupo/geopm.git geopm-rpm
fi
cd geopm-rpm
git checkout cmcantalupo-theta-spec-file
git reset --hard origin/cmcantalupo-theta-spec-file

git clean -dfx
rm -rf ~/rpmbuild/
./autogen.sh
./configure
make rpm-theta
```

cmcantalupo, 2018-08-16:
Spec file and make target have been integrated here:

https://github.com/geopm/geopm/commit/9506137f16566ab19fa6f2159524523ccdb1c47b

closed 2018-08-16 by cmcantalupo