geopm/geopm #336: Debug ControlMessage::wait() errors on Theta

bgeltz, 2018-08-15:
Very rarely, I'm seeing these messages in my job logs on Theta at job start:

```
Error: <geopm> Runtime error: ControlMessage::wait(): Timed out waiting for status 1: at geopm/src/ControlMessage.cpp:87             
Error: <geopm> Runtime error: ControlMessage::wait(): Timed out waiting for status 4: at geopm/src/ControlMessage.cpp:87
```

Once this happens, the job hangs indefinitely.  It will eventually be killed by the resource manager when it times out.  The app appears to run to completion, but the Controller is hung.

The last job I have that reproduced this problem is 262202.  Here are the relevant docs:

```
/projects/intel/geopm-home/shared/output/262202.output
/projects/intel/geopm-home/shared/output/262202.error
/projects/intel/geopm-home/shared/output/262202.cobaltlog
/projects/intel/geopm-home/shared/output/262202-nekbone-256node-1rank-62thread-2018-08-15_1011```

bgeltz, 2018-08-17:
The following jobs have had this issue to date:

257494.error
257781.error
257830.error
260629.error
261980.error
262202.error
262604.error
262854.error
262858.error

bgeltz, 2018-08-17:
The following jobs have had the issue to date:

257494
261980
262202
262604
262854
262858

bgeltz, 2018-08-18:
```grep "Timed out waiting for status 1" *error | cut -d: -f1 | xargs ls -la```

bgeltz, 2018-08-20:
New failures over the weekend:
262863
262877
262884
262945
262946

bgeltz, 2018-08-30:
These issues persist, but only for Nekbone.  @dianarg is not having problems with AMG or MiniFE.

I'm trying to get this up in DDT.  I can't make progress on this presently because of #344.  I must use the pthread launch method because of DDT.  DDT hangs at launch on Theta with the process launch method.

bgeltz, 2018-09-04:
The issues around launching in pthread mode have been fixed on Mr. Fusion.  The issue about OMP process masks being zeroed (#346) remains.  It must be resolved before DDT can be utilized with the pthread launch mechanism.

bgeltz, 2018-09-13:
Turns out DDT is a no-go since this is a problem that occurs within MPI_Init.  printf is the only way to go...
https://github.com/cmcantalupo/geopm/commit/46b2a96
https://github.com/cmcantalupo/geopm/commit/66a9e266c6c9cd1017a6c34a0d9742b3fba76a3a
https://github.com/cmcantalupo/geopm/commit/c018e48dc7f5fa4d58234e692afe4cff93beb102

bgeltz, 2018-09-19:
Chris and I have been suspecting there's some sort of race that happens with the creation of the SharedMemory region that's used between the app and the Controller.  What should happen is that the Controller starts up, creates a shmem region, calls ftruncate() to grow the region, then proceed onward.

The app should first wait until the region is created, then wait for the region to be grown (i.e. fstat() should not return size 0), then proceed onward.  After some digging yesterday, it appears as though ftruncate is *not* atomic and simply having the app wait for fstat() to return a non-zero size is not enough to ensure the region is ready.  

The ftruncate() code on the Controller side is here:
https://github.com/geopm/geopm/blob/bdddc85e03dcd0a1a4073ed63bd581ed885ae95a/src/SharedMemory.cpp#L59-L65

The fstat() code on the app size is here:
https://github.com/geopm/geopm/blob/bdddc85e03dcd0a1a4073ed63bd581ed885ae95a/src/SharedMemory.cpp#L164-L170

I've been testing a workaround that sleeps for a period of time after the fstat() call returns non-zero.  It's in commit 6325a82a9662ec1b3173d587818a65016a66a434, presently on my [bgeltz-theta](/bgeltz/geopm/tree/bgeltz-theta) branch.  It appears to resolve the race in my long job that ran last night.

If this is indeed the culprit, then we need to introduce some sort of lock bit/ready value that the Controller writes into the shmem region.  That way, the app would continue to wait until this known value appears in the region.

cmcantalupo, 2018-09-21:
https://review.gerrithub.io/#/c/geopm/geopm/+/426122/

closed 2018-09-21 by cmcantalupo