geopm/geopm #346: OMP process masks zeroed

bgeltz, 2018-09-04:
Related to #344, the process mask being returned from geopm_sched_woomp() is zeroed out on Theta.  This is leading to failures in pthread_create().

Here's the entry point in the PMPI wrapper:
https://github.com/geopm/geopm/blob/dev/src/geopm_pmpi.c#L226

I've put some print's in the code to illustrate the problem:
177ee65034ac8b81a2fe90441c94b80d7163a028

Here's the resulting output:
```
thread mask1 is 0
thread mask2 is 0
thread mask3 is 0
thread mask4 is 0
```

Then a short time later:
```
Error: Invalid argument: Controller::pthread(): pthread_create() failed: at geopm/src/Kontroller.cpp:306
```

Presently the Controller cannot launch with this issue.  This blocks debug with DDT.

Reproducing this on theta requires ```MPICH_MAX_THREAD_SAFETY=multiple``` in your job script.  I'm currently debugging this in two phases:
1. Invoke ```geopmaprun``` to get the full ```aprun``` commandline to be executed.  Then stop the job.
2. With the full ```aprun``` commandline in hand, paste it into the script, comment out the call to ```geopmaprun``` and then add ```/soft/debuggers/ddt/bin/ddt --connect``` just before ```aprun```.

The run script I've been using is here:
```/projects/intel/geopm-home/bgeltz/hack_scripts/app_debug_nekbone_2.sh```

bgeltz, 2018-09-07:
https://review.gerrithub.io/c/geopm/geopm/+/424896

cmcantalupo, 2018-09-08:
Actually should have used CPU_COUNT_S(CPU_ALLOC_SIZE(num_cpu), cpu_set).

cmcantalupo, 2018-09-10:
https://review.gerrithub.io/#/c/geopm/geopm/+/425164/

commit 1713d43c (2018-09-08): Fix handling of bad OMP masks.

- Fixes #346.

Change-Id: Ia5a5138d396599da40881d9783d7ea32f032541e
closed 2018-09-14 by cmcantalupo