geopm/geopm #363: geopmread broken for certain signals

bgeltz, 2018-09-21:
```
$ geopmread POWER_PACKAGE package 0
Error: cannot read signal: <geopm> Invalid argument: PlatformIO::signal_domain_type(): signal name "REGION_ID#" not found: at geopm/src/PlatformIO.cpp:151
```

```
$ geopmread POWER_DRAM board_memory 0
Error: cannot read signal: <geopm> Invalid argument: PlatformIO::signal_domain_type(): signal name "REGION_ID#" not found: at geopm/src/PlatformIO.cpp:151
```

dianarg, 2018-09-21:
Possibly related/duplicate of #266

dianarg, 2018-10-31:
Once #266 is implemented, geopmread can have special handling for POWER_* signals that collects multiple samples over a period of time.  The last sample will be the most accurate.

commit 41e7b97d (2018-11-07): Read multiple samples for a short time in geopmread.

- Added to support POWER_* as signals that give meaningful results.
- Fixes #363.

Change-Id: I33580391fd83a082b7fe8d0e1c5e3875d71fb110
closed 2018-11-07 by cmcantalupo