geopm/geopm #413: geopmanalysis requires SLURM or COBALT node ENV variables

lhlawson, 2018-11-06:
When checking launch options in the main function of analysis.py, SLURM_NNODES and COBALT_NODEFILE are checked to determine num_nodes even if SLURM isn't being used.  
	
If neither env var is found num_nodes is set to -1 which results in a RuntimeError (if num_nodes != the command line num_nodes).  
	
Run command: 
	GEOPM_RM=IMPI geopmanalysis freq_sweep  -N1 -n1 --min-freq 1e9 --max-freq 2e9  -- ~/build/bin/geopmbench
Error: 
	RuntimeError: Launch must be made inside of a job allocation and application must run on all allocated nodes.

dianarg, 2018-12-04:
I think this requirement has been removed.  It is now the responsibility of users to make sure they use the same set of nodes from an allocation.

dianarg, 2020-03-18:
Not sure if it's our responsibility to tell users they should use the same set of nodes for comparisons.  Such advice could maybe go in the same place as #691

dianarg, 2020-04-01:
This message has been removed.  The documentation needs to be added.