geopm/geopm #416: geopmbench 1.0 big-o for stream is too long

cmcantalupo, 2018-11-13:
The stream model region with big-o of 1.0 runs for *much* longer than 1 second.

Related to #382

maiterth, 2018-11-13:
Small update: the configure script looks for HAVE_ICPC to enable mkl.
If CC and CXX are not explicitly set to icc and icpc, that path for compilation is not chosen even if the rest of geopm is built fine. This entails non-optimal performance. 
tutorial_6 using the same code path as geopmbench runs fine since it is built using the tutorial_build_intel.sh scripts. This might not be obvious when testing.

cmcantalupo, 2018-11-14:
To be clear the issue is actually the performance of DGEMM when not using the intel toolchain to build geopmbench.  It is not related to the calibration of the stream model region.

closed 2018-11-30 by cmcantalupo