geopm/geopm #448: Ambiguous error when running an agent with a missing policy value

lhlawson, 2018-12-10:
If a key/value pair is missing from the policy file for an agent the error provided indicates a signal of that key name is not found.  

Example:  Deleting LOW_THRESHOLD from geopm/tutorial/agent/example_policy.json results in 
- Error: <geopm> Invalid argument: ManagerIOSampler::read_batch(): Signal "LOW_THRESHOLD" not found.: at geopm/src/ManagerIO.cpp:296

This error is usable but somewhat ambiguous and may cause confusion.  

To Reproduce: 
Remove the LOW_THRESHOLD key/value pair from the example_policy.json file and run agent_tutorial.sh

dianarg, 2020-03-31:
Pretty sure this was fixed with fix for invalid policy files

dianarg, 2020-04-01:
This has been fixed.

closed 2020-04-01 by dianarg