geopm/geopm #470: Run integration tests on Quartz

cmcantalupo, 2019-01-08:


bgeltz, 2019-04-10:
I've been looking at this on SKX.  The only test that isn't related to RAPL that I'm having issues with is:

```
======================================================================
FAIL: test_runtime_regulator (__main__.TestIntegration)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "./geopm_test_integration.py", line 511, in test_runtime_regulator
    self.assertNear(df['region_runtime'], expected_region_runtime[region_name], epsilon=0.001)
  File "./geopm_test_integration.py", line 181, in assertNear
    self.fail('The fractional difference between {a} and {b} is greater than {epsilon}.  {msg}'.format(a=a, b=b, epsilon=epsilon, msg=msg))
AssertionError: The fractional difference between 1.001918357 and 1.0 is greater than 0.001.
```

I’m trying to figure out why this test was written to use the “region_runtime” field in the trace instead of getting the progress entry and exit row from a single execution of the region it’s working on and diffing the “time” fields (other tests do that).  I thought it had something to do with the “region_runtime” being calculated by the runtime regulator or similar, but I cannot recall.

What I’m seeing right now, is that the test is blowing up on the very first comparison of the “region_runtime” trace data.  It should be very, very close to 1 second.

I put a break right before the assertNear so I could inspect the data.  I’m seeing:

```
>>> filtered_df[:10]
          time  region_progress  region_runtime
0     0.037690              0.0        0.000000
200   1.037643              1.0        1.001918
201   1.537872              0.0        1.002391
401   2.537817              1.0        1.002391
402   3.038077              0.0        1.001123
603   4.042961              1.0        1.002404
604   4.543216              0.0        1.002404
804   5.543140              1.0        1.002404
805   6.043372              0.0        1.002404
1006  7.048309              1.0        1.002404
```

So 1.001918 is too far out of range for the assertNear.  If I do a simple 1.037643 -  0.037690 = 0.9999530000000001 which is fine for the assertNear.

Open ?’s:
    What causes the region_runtime field in the trace to update?
    Why are there a bunch of repeated numbers in that field?

bgeltz, 2019-04-10:
I just finished resurrecting the ProfileTracer for this debug:
https://github.com/bgeltz/geopm/commit/545e5dbe9d80469561d545bc1d41c68270c9ffd6

bgeltz, 2019-04-10:
Current dataset:

```
>>> filtered_df[:10]
          time  region_progress  region_runtime
0     0.052491              0.0        0.000000
200   1.052409              1.0        1.001077
201   1.552627              0.0        1.001558
401   2.552663              1.0        1.002389
402   3.052919              0.0        1.002389
603   4.057792              1.0        1.002405
604   4.558066              0.0        1.002405
804   5.557980              1.0        1.002404
805   6.058202              0.0        1.002404
1006  7.063137              1.0        1.002405
>>> region_name
'sleep'
```

Trying to hone in on the first recorded region_runtime, ```1.001077```.  I ran this test with profile tracing and pulled out the first entries for all 4 ranks for the sleep region:
```
rank|region_hash|region_hint|timestamp|progress
3|0x00000000536c798f|0x0000000100000000|4.8119082000000001e-02|0.0000000000000000e+00
2|0x00000000536c798f|0x0000000100000000|4.8119358000000001e-02|0.0000000000000000e+00
1|0x00000000536c798f|0x0000000100000000|4.8119031000000000e-02|0.0000000000000000e+00
0|0x00000000536c798f|0x0000000100000000|4.8119180000000004e-02|0.0000000000000000e+00
3|0x00000000536c798f|0x0000000100000000|1.0491956830000000e+00|1.0000000000000000e+00
2|0x00000000536c798f|0x0000000100000000|1.0493008920000000e+00|1.0000000000000000e+00
1|0x00000000536c798f|0x0000000100000000|1.0495743490000000e+00|1.0000000000000000e+00
0|0x00000000536c798f|0x0000000100000000|1.0496773150000001e+00|1.0000000000000000e+00
```

Doing some math to get each ranks runtime:
```
>>> rr = pandas.Series([], dtype='double')
>>> rr[0] = 1.0496773150000001e+00 - 4.8119180000000004e-02
>>> rr[1] = 1.0495743490000000e+00 - 4.8119031000000000e-02
>>> rr[2] = 1.0493008920000000e+00 - 4.8119358000000001e-02
>>> rr[3] = 1.0491956830000000e+00 - 4.8119082000000001e-02
>>> rr.describe()
count    4.000000
mean     1.001318
std      0.000226
min      1.001077
25%      1.001155
50%      1.001318
75%      1.001481
max      1.001558
dtype: float64
```

So even though the agg function for this signal is MAX, the MIN is being recorded in the trace.

bgeltz, 2019-04-10:
Next, analyzing the repeated values.  Trying to calculate the region runtime for rows 603, 604, and 804:

(Same dataset as above)
```
>>> filtered_df[:10]
          time  region_progress  region_runtime
0     0.052491              0.0        0.000000
200   1.052409              1.0        1.001077
201   1.552627              0.0        1.001558
401   2.552663              1.0        1.002389
402   3.052919              0.0        1.002389
603   4.057792              1.0        1.002405
604   4.558066              0.0        1.002405
804   5.557980              1.0        1.002404
805   6.058202              0.0        1.002404
1006  7.063137              1.0        1.002405
>>> region_name
'sleep'
```

Profile trace data:
```
rank|region_hash|region_hint|timestamp|progress
3|0x00000000536c798f|0x0000000100000000|3.0524343140000001e+00|0.0000000000000000e+00
2|0x00000000536c798f|0x0000000100000000|3.0524343859999998e+00|0.0000000000000000e+00
1|0x00000000536c798f|0x0000000100000000|3.0524342060000000e+00|0.0000000000000000e+00
0|0x00000000536c798f|0x0000000100000000|3.0524343240000000e+00|0.0000000000000000e+00

3|0x00000000536c798f|0x0000000100000000|4.0548393249999997e+00|1.0000000000000000e+00
2|0x00000000536c798f|0x0000000100000000|4.0548387269999999e+00|1.0000000000000000e+00
1|0x00000000536c798f|0x0000000100000000|4.0548378370000000e+00|1.0000000000000000e+00
0|0x00000000536c798f|0x0000000100000000|4.0548384940000002e+00|1.0000000000000000e+00

3|0x00000000536c798f|0x0000000100000000|4.5548614780000003e+00|0.0000000000000000e+00
2|0x00000000536c798f|0x0000000100000000|4.5548614530000000e+00|0.0000000000000000e+00
1|0x00000000536c798f|0x0000000100000000|4.5548613979999999e+00|0.0000000000000000e+00
0|0x00000000536c798f|0x0000000100000000|4.5548612999999998e+00|0.0000000000000000e+00

3|0x00000000536c798f|0x0000000100000000|5.5571731389999997e+00|1.0000000000000000e+00
2|0x00000000536c798f|0x0000000100000000|5.5572653220000001e+00|1.0000000000000000e+00
1|0x00000000536c798f|0x0000000100000000|5.5572648889999998e+00|1.0000000000000000e+00
0|0x00000000536c798f|0x0000000100000000|5.5572649690000002e+00|1.0000000000000000e+00
```
Maths on the top set:
```>>> rr = pandas.Series([], dtype='double')
>>> rr[0] = 4.0548384940000002e+00 - 3.0524343240000000e+00
>>> rr[1] = 4.0548378370000000e+00 - 3.0524342060000000e+00
>>> rr[2] = 4.0548387269999999e+00 - 3.0524343859999998e+00
>>> rr[3] = 4.0548393249999997e+00 - 3.0524343140000001e+00
>>> rr
0    1.002404
1    1.002404
2    1.002404
3    1.002405
dtype: float64
>>> rr.describe()
count    4.000000e+00
mean     1.002404e+00
std      5.689495e-07
min      1.002404e+00
25%      1.002404e+00
50%      1.002404e+00
75%      1.002405e+00
max      1.002405e+00
dtype: float64
```
Maths on the bottom set:
```
>>> rr = pandas.Series([], dtype='double')
>>> rr[0] = 5.5572649690000002e+00 - 4.5548612999999998e+00
>>> rr[1] = 5.5572648889999998e+00 - 4.5548613979999999e+00
>>> rr[2] = 5.5572653220000001e+00 - 4.5548614530000000e+00
>>> rr[3] = 5.5571731389999997e+00 - 4.5548614780000003e+00
>>> rr
0    1.002404
1    1.002403
2    1.002404
3    1.002312
dtype: float64
>>> rr.describe()
count    4.000000
mean     1.002381
std      0.000046
min      1.002312
25%      1.002381
50%      1.002404
75%      1.002404
max      1.002404
dtype: float64
```

bgeltz, 2019-04-11:
Resolved with:
https://review.gerrithub.io/c/geopm/geopm/+/450872

closed 2019-04-17 by cmcantalupo