geopm/geopm #481: Controller handshake failing

bgeltz, 2019-01-14:
I'm currently bumping into this on Theta with the tip of dev.  I'm reproducing this on as little as 2 nodes with 6 ranks per node and 10 threads.  The COBALT script is here:
/projects/intel/geopm-home/shared/scripts/app_hacc_test.sh

When I run that I see:
/projects/intel/geopm-home/shared/output/303573.output:
```
...
GEOPM_AGENT=power_balancer GEOPM_PROFILE=205 GEOPM_POLICY=power_balancer_205_1_policy.json LD_DYNAMIC_WEAK=true GEOPM_REPORT=hacc_power_balancer_205_1_report OMP_PROC_BIND=true GEOPM_PROFILE_TIMEOUT=300 GEOPM_CTL=process MV2_ENABLE_AFFINITY=0 KMP_AFFINITY=disabled KMP_WARNINGS=FALSE aprun -N 7 -n 14 --cpu-binding 1:4-13:14-23:24-33:34-43:44-53:54-63 -- numactl -m 0 /projects/intel/geopm-home/bgeltz/apps/HACC_v1.5_Theta/src/cpu/theta/hacc_tpm indat cmbM000.tf m000 INIT ALL_TO_ALL -w -R -N 512 -t 2x2x3                               
                                                                                                       
Partition 3D: [2:2:3]                                      
Threads per process: 10
Allocated heap: 6.05 MB (large) + 1.69 MB (small) used, 0.25 MB unused
Done loading parameters.
Initializing redistribution using a pencil layout on 14 ranks.
distribution 1D: [14:1:1]
Application 10622833 exit codes: 134
Application 10622833 exit signals: Killed
Application 10622833 resources: utime ~36s, stime ~183s, Rss ~9220, inblocks ~2720, outblocks ~0
...
```

/projects/intel/geopm-home/shared/output/303573.error:
```
...
+ OMP_NUM_THREADS=10                                                                                                                  
+ geopmlaunch aprun -N 6 -n 12 --geopm-agent=power_governor --geopm-ctl=process --geopm-policy=power_governor_205_1_policy.json --geo$
m-report=hacc_power_governor_205_1_report --geopm-profile=205 --geopm-timeout=300 -- numactl -m 0 /projects/intel/geopm-home/bgeltz/a$
ps/HACC_v1.5_Theta/src/cpu/theta/hacc_tpm indat cmbM000.tf m000 INIT ALL_TO_ALL -w -R -N 512 -t 2x2x3                                 
Warning: <geopm> Controller handshake failed, running without geopm.                                                                  
No such file or directory                                                                                                             
Rank 13 [Mon Jan 14 14:26:43 2019] [c0-0c0s2n3] Fatal error in PMPI_Comm_rank: Invalid communicator, error stack:                     
PMPI_Comm_rank(110): MPI_Comm_rank(MPI_COMM_NULL, rank=0x100e614) failed                                                              
PMPI_Comm_rank(67).: Null communicator                                                                                                
Rank 12 [Mon Jan 14 14:26:43 2019] [c0-0c0s2n3] Fatal error in PMPI_Comm_rank: Invalid communicator, error stack:                     
PMPI_Comm_rank(110): MPI_Comm_rank(MPI_COMM_NULL, rank=0x100e614) failed                                                              
PMPI_Comm_rank(67).: Null communicator                                                                                                
_pmiu_daemon(SIGCHLD): [NID 00011] [c0-0c0s2n3] [Mon Jan 14 14:26:43 2019] PE RANK 12 exit signal Aborted                             
[NID 00011] 2019-01-14 22:26:43 Apid 10622970: initiated application termination
<geopmpy.launcher> Command 'aprun -N 7 -n 14 --cpu-binding 1:4-13:14-23:24-33:34-43:44-53:54-63 -- numactl -m 0 /projects/intel/geopm$
home/bgeltz/apps/HACC_v1.5_Theta/src/cpu/theta/hacc_tpm indat cmbM000.tf m000 INIT ALL_TO_ALL -w -R -N 512 -t 2x2x3' returned non-zer$
 exit status 137
...
```

bgeltz, 2019-01-14:
Note that I am able to run tutorial_2 successfully in the debug queue.

bgeltz, 2019-01-15:
This was resolved by recompiling the target app.

bgeltz, 2019-01-15:
I was able to run successfully overnight by simply recompiling the target app as I said in my last comment.  My last run was HACC on 256 nodes on Theta with trace files disabled.  I'm testing that again now with a different set of RAPL budgets.

If this is indeed the case, I'd love to bottom out on why this happened.  It might be time to version the app/controller interface and do some sanity checks when someone tries to connect...

cmcantalupo, 2019-02-04:
https://github.com/geopm/geopm/commit/6f868b3c6111f17f280a406c6a036fda6e90a81b

bgeltz, 2019-02-07:
It's back:
```
309932.error:
    Warning: <geopm> Controller handshake failed, running without geopm.
    No such file or directory
```

closed 2020-03-31 by dianarg