geopm/geopm #544: Release version v1.0.0

cmcantalupo, 2019-03-06:


dianarg, 2019-03-15:
Need to merge:
https://review.gerrithub.io/c/geopm/geopm/+/445982

cmcantalupo, 2019-04-03:
Pre rc3 tasks:
- [ ] Update ChangeLog based on git log. - @cmcantalupo
- [ ] Update README dates and contact info. @cmcantalupo
- [ ] Update doxygen, man pages and code coverage public web pages. @cmcantalupo 
- [ ] Make sure OBS is building the dev branch correctly. @cmcantalupo
- [ ] Adding the release tag @cmcantalupo 

Post rc3 tasks:
- [ ] Push tarball to python PIP test server and make sure it builds properly there.
- [ ] Make sure static analysis is passing (IPLDT, klocwork, protex).
- [ ] Update coverage numbers.
- [ ] Run integration tests on mr-fusion, quartz, and theta.
- [ ] Make sure tutorials build and execute.
- [ ] Make sure OMPT works (not configured on in most of our tests).

closed 2019-04-17 by cmcantalupo