geopm/geopm #580: Wrong affinity parsing when used with --disable-procfs

sbtrasi, 2019-04-16:
MPI rank's affinity mask seems to be parsed incorrectly when GeoPM is built using --disable-procfs

$ ./configure --disable-procfs
$ make 
$ make install
$ ./run_on_linux.sh 3
Starting GeoPM Controller...
Starting Application...
Warning: <geopm> Controller handshake failed, running without geopm.
<geopm> MPI ranks are not affinitized to distinct CPUs
MPI_COMM_WORLD size: 2
Beginning loop of 500 iterations.

dianarg, 2019-04-16:
Thanks for the report @sbtrasi.  Can you post the contents of the run_on_linux.sh script?

sbtrasi, 2019-04-16:
#!/bin/bash

RANKS=2
THREADS=46
POLICY="governed"
AGENT="power_governor"
HOST=skl-9

Help()
{
        echo ""
        echo "Help:"
        echo "./run_on_mos.sh <tutorial number> <power policy>"
        echo ""
        echo "<power policy> -> g for governed"
        echo "                  b for balanced"
        exit 0
}

if [[ -z $1 || ! -e ./tutorial_$1 ]]; then
        Help
else
        if [[ $2 == "b" ]]; then
                POLICY="balanced"
                AGENT="power_balancer"
        fi
fi

(echo "Starting GeoPM Controller..."; mpirun --hosts $HOST -env I_MPI_PIN_DOMAIN=[0x1] -ppn 1 -np 1 geopmctl) &

export PSM2_MQ_RNDV_HFI_THRESH=200000
export PSM2_MQ_EAGER_SDMA_SZ=65536
export PSM2_MQ_RNDV_HFI_WINDOW=4194304
export PSM2_RCVTHREAD=0
export I_MPI_PIN=on
export I_MPI_FALLBACK=0
export I_MPI_HARD_FINALIZE=1
export I_MPI_TMI_PROVIDER=psm2
export I_MPI_FABRICS=shm:tmi
export I_MPI_PIN_PROCESSOR_EXCLUDE_LIST=0,24,48,72
export I_MPI_PIN_DOMAIN=46
export KMP_AFFINITY=compact
export KMP_HW_SUBSET=2t

export LD_PRELOAD=/nfshome/bhatshar/geopm/build/lib/libgeopm.so
export LD_DYNAMIC_WEAK=true
export GEOPM_REPORT=tutorial_${1}_${POLICY}_report
export GEOPM_TRACE=tutorial_${1}_${POLICY}_trace
export GEOPM_AGENT=${AGENT}
export GEOPM_POLICY=tutorial_${POLICY}_policy.json

echo "Starting Application..."
mpirun --hosts $HOST -ppn ${RANKS} -np ${RANKS} -env OMP_NUM_THREADS=${THREADS} ./tutorial_${1}

closed 2022-03-23 by cmcantalupo