geopm/geopm #597: Set GEOPM_TIMEOUT to 0 if not launching geopm

dianarg, 2019-05-03:
If a user links their application with libgeopm but runs the application without geopm, they will still get a 30 second timeout waiting for the controller.  This should be set up so that there is no wait when running without geopm, and increase the timeout to the default 30s when using the launcher or when GEOPM_CTL/GEOPM_REPORT/GEOPM_PROFILE are set.

dianarg, 2019-11-22:
Related to #499

sidjana, 2020-03-12:
Set the TIMEOUT to 0 by default. Have the launching script be responsible for setting to a different value when GEOPM isn't used.

bgeltz, 2023-07-28:
The implementation around this has changed.  The application no longer waits on the specified GEOPM_TIMEOUT.  Only the controller will wait if it is set.

closed 2023-07-28 by bgeltz