geopm/geopm #599: Use agg:sun for geopmread POWER_PACKAGE_TDP board

bgeltz, 2019-05-06:
Currently issuing geopmread for the TDP signal for the board domain returns the value for a single socket.  On a multi socket system, requesting this at the board domain should sum the value for all sockets.

This will aid in policy construction for power balancer, etc. as they require limits set for the entire board.

bgeltz, 2019-06-19:
Think about `POWER_PACKAGE_TDP` vs. `POWER_TDP` as a PlatformIO alias to solve this.  It's not avoidable for low-level implementations, but should work for higher level aliases.

dianarg, 2019-06-19:
If sum() is appropriate for the aggregation function for POWER_PACKAGE_TDP, it makes sense to change POWER_PACKAGE_MIN and POWER_PACKAGE_MAX also.

commit 2522cf83 (2019-10-29): Change board power min/max/tdp to use sum aggregation

- Fixes #599

Change-Id: I2f7b18c0d4ee23f72e481f7129623169c148e0d3
closed 2019-10-29 by cmcantalupo