geopm/geopm #605: Update PlatformTopo C++ man page

vysocky, 2019-05-16:
[Man page of PlatformTopo](https://geopm.github.io/man/GEOPM_CXX_MAN_PlatformTopo.3.html) refers to `geopm::PlatformTopo::domain_cpus(...)`, which is not a method of the class any more.

```
using geopm::PlatformTopo;

int main() {
    PlatformTopo &topo = geopm::platform_topo();
...
    for (int pkg_idx = 0; pkg_idx < num_pkgs; ++pkg_idx) {
        std::cout << "CPUs on package " << pkg_idx << ": ";
        std::set<int> cpus = topo.domain_cpus(GEOPM_DOMAIN_PACKAGE, pkg_idx);
        for(auto pcpu : cpus) {
            std::cout << pcpu << " ";
        }
        std::cout << std::endl;
    }
```
Also geopm::platform_topo() return type is now  `const`.

#454

cmcantalupo, 2019-05-17:
Thank for the feedback @vysocky!  The domain_cpus() became private and was replaced by the more general domain_nested() method.  I will update the documentation to reflect this change soon.

cmcantalupo, 2019-05-20:
https://review.gerrithub.io/c/geopm/geopm/+/455130

commit 038e59c0 (2019-05-22): Fix PlatformTopo C++ man page

- Change references to domain_cpus() to replacement API domain_nested().
- Change topo reference to const in man page example.
- Fixes #605

Change-Id: I3bbb7dbf8da719844998f6658caac1a4048799e5
closed 2019-05-22 by cmcantalupo