geopm/geopm #614: Launcher issues on CoolMUC

cmcantalupo, 2019-05-28:
Appears to be an affinity problem with the launcher on CoolMUC.  Controller complains that MPI process is not bound to distinct CPUs.  

```
GEOPM_TRACE=tutorial_0_trace GEOPM_TIMEOUT=5 GEOPM_PROFILE= LD_DYNAMIC_WEAK=true GEOPM_REPORT=tutorial_0_report OMP_PROC_BIND=true GEOPM_CTL=process LD_PRELOAD=libgeopm.so MV2_ENABLE_AFFINITY=0 KMP_AFFINITY=disabled KMP_WARNINGS=FALSE I_MPI_PIN_DOMAIN=[0x10000000000000000,0x2,0x4,0x8,0x10,0x20,0x40,0x80,0x100,0x200,0x400,0x800,0x1000,0x2000,0x4000,0x8000,0x10000,0x20000,0x40000,0x80000,0x100000,0x200000,0x400000,0x800000,0x1000000,0x2000000,0x4000000,0x8000000,0x10000000,0x20000000,0x40000000,0x80000000,0x100000000,0x200000000,0x400000000,0x800000000,0x1000000000,0x2000000000,0x4000000000,0x8000000000,0x10000000000,0x20000000000,0x40000000000,0x80000000000,0x100000000000,0x200000000000,0x400000000000,0x800000000000,0x1000000000000,0x2000000000000,0x4000000000000,0x8000000000000,0x10000000000000,0x20000000000000,0x40000000000000,0x80000000000000,0x100000000000000,0x200000000000000,0x400000000000000,0x800000000000000,0x1000000000000000,0x2000000000000000,0x4000000000000000,0x8000000000000000] mpiexec.hydra -ppn 64 -n 128 ./tutorial_0

MPI_COMM_WORLD size: 126
Sleeping for five seconds
Warning: <geopm> Controller handshake failed, running without geopm.
<geopm> MPI ranks are not affinitized to distinct CPUs
Error: <geopm> Runtime error: ControlMessageImp::wait(): mpp3r03c01s04 : is_ctl=1 : is_writer=1 : Timed out waiting for status 2: at geopm/src/ControlMessage.cpp:111
Error: <geopm> Runtime error: ControlMessageImp::wait(): mpp3r03c01s09 : is_ctl=1 : is_writer=1 : Timed out waiting for status 2: at geopm/src/ControlMessage.cpp:111

```

maiterth, 2019-06-05:
There are three things:
- The tutorial do not have impi as a launcher option. Should this be added?
- The tutotiral_env sets srun as default even if MPIEXEC is chosen. GEOPM_LAUNCHER has to be set to a not matching paramter, but has to be set.
- When not using the complete node the launcher fails. (Examples are using 10 processes on 2 nodes, but in this case the communication split is wrong. Using -ppn 64 and -n 128 is working (KNL). when setting I_MPI_PIN=1 and I_MPI_PIN_DOMAIN=core. (default of I_MPI_PIN_DOMAIN=[0x2,...,0x800...] triggers warnings.) 

Still investigating.

bgeltz, 2019-06-12:
For anyone trying to get builds going on CoolMUC3, I had to add:
`export CPATH=${MKL_INCDIR}`

Before I built in order to work around this being missing from the Intel and MKL modules.

bgeltz, 2019-06-12:
I'm seeing unit test failures around affinity:
```FAIL: scripts/test/pytest_links/TestAffinity.test_1rank_no_env_threads
FAIL: scripts/test/pytest_links/TestAffinity.test_1rank_no_env_threads_noht
FAIL: scripts/test/pytest_links/TestAffinity.test_2rank_no_env_threads
FAIL: scripts/test/pytest_links/TestAffinity.test_2rank_no_env_threads_noht
FAIL: scripts/test/pytest_links/TestAffinity.test_3rank_no_env_threads
FAIL: scripts/test/pytest_links/TestAffinity.test_3rank_no_env_threads_noht
```

It turns out one of the auto loaded modules on CoolMUC3 sets OMP_NUM_THREADS:
```$ module show  admin/1.0 | grep OMP_NUM_THREADS
setenv          OMP_NUM_THREADS 1
```
Adding `unset OMP_NUM_THREADS` to the bottom of my .bashrc made my unit tests pass.

bgeltz, 2019-06-13:
It sure looks like something is broken with IMPI pinning:
```
$ OMP_NUM_THREADS=4 geopmlaunch impi -n 4 -ppn 4 ./print_affinity

mpiexec.hydra -ppn 1 -n 1 geopmread --cache

GEOPM_CTL=process LD_DYNAMIC_WEAK=true GEOPM_REPORT=geopm.report OMP_PROC_BIND=true GEOPM_PROFILE= LD_PRELOAD=libgeopm.so MV2_ENABLE_AFFINITY=0 KMP_AFFINITY=disabled KMP_WARNINGS=FALSE I_MPI_PIN_DOMAIN=[0x2,0xf000000000000,0xf0000000000000,0xf00000000000000,0xf000000000000000] mpiexec.hydra -ppn 5 -n 5 ./print_affinity

Warning: <geopm> Controller handshake failed, running without geopm.
<geopm> MPI ranks are not affinitized to distinct CPUs
Rank: 0 cgroup CPUs: [0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,65,66,67,68,69,70,71,72,73,74,75,76,77,78,79,80,81,82,83,84,85,86,87,88,89,90,91,92,93,94,95,96,97,98,99,100,101,102,103,104,105,106,107,108,109,110,111,112,113,114,115,116,117,118,119,120,121,122,123,124,125,126,127,128,129,130,131,132,133,134,135,136,137,138,139,140,141,142,143,144,145,146,147,148,149,150,151,152,153,154,155,156,157,158,159,160,161,162,163,164,165,166,167,168,169,170,171,172,173,174,175,176,177,178,179,180,181,182,183,184,185,186,187,188,189,190,191,192,193,194,195,196,197,198,199,200,201,202,203,204,205,206,207,208,209,210,211,212,213,214,215,216,217,218,219,220,221,222,223,224,225,226,227,228,229,230,231,232,233,234,235,236,237,238,239,240,241,242,243,244,245,246,247,248,249,250,251,252,253,254,255] omp CPUs: [24,27,157,31]
Error: <geopm> Runtime error: ControlMessageImp::wait(): mpp3r03c05s03 : is_ctl=1 : is_writer=1 : Timed out waiting for status 2: at geopm/src/ControlMessage.cpp:111
```
I can reproduce on mr-fusion:
```
$ OMP_NUM_THREADS=4 geopmlaunch impi -n 4 -ppn 4 ./print_affinity

mpiexec.hydra -ppn 1 -n 1 geopmread --cache

GEOPM_CTL=process LD_DYNAMIC_WEAK=true GEOPM_REPORT=geopm.report OMP_PROC_BIND=true GEOPM_PROFILE= LD_PRELOAD=libgeopm.so MV2_ENABLE_AFFINITY=0 KMP_AFFINITY=disabled KMP_WARNINGS=FALSE I_MPI_PIN_DOMAIN=[0x2,0xf000000000000,0xf0000000000000,0xf00000000000000,0xf000000000000000] mpiexec.hydra -ppn 5 -n 5 ./print_affinity

Warning: <geopm> Controller handshake failed, running without geopm.
<geopm> MPI ranks are not affinitized to distinct CPUs
Rank: 0 cgroup CPUs: [0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,65,66,67,68,69,70,71,72,73,74,75,76,77,78,79,80,81,82,83,84,85,86,87,88,89,90,91,92,93,94,95,96,97,98,99,100,101,102,103,104,105,106,107,108,109,110,111,112,113,114,115,116,117,118,119,120,121,122,123,124,125,126,127,128,129,130,131,132,133,134,135,136,137,138,139,140,141,142,143,144,145,146,147,148,149,150,151,152,153,154,155,156,157,158,159,160,161,162,163,164,165,166,167,168,169,170,171,172,173,174,175,176,177,178,179,180,181,182,183,184,185,186,187,188,189,190,191,192,193,194,195,196,197,198,199,200,201,202,203,204,205,206,207,208,209,210,211,212,213,214,215,216,217,218,219,220,221,222,223,224,225,226,227,228,229,230,231,232,233,234,235,236,237,238,239,240,241,242,243,244,245,246,247,248,249,250,251,252,253,254,255] omp CPUs: [128,2,69,67]
Error: <geopm> Runtime error: ControlMessageImp::wait(): mr-fusion2 : is_ctl=1 : is_writer=1 : Timed out waiting for status 2: at geopm/src/ControlMessage.cpp:112
```
However, it runs just fine on Chameleon:
```
$ FI_PROVIDER=tcp OMP_NUM_THREADS=4 geopmlaunch impi -n 4 -ppn 4 ./print_affinity

mpiexec.hydra -ppn 1 -n 1 geopmread --cache

GEOPM_CTL=process LD_DYNAMIC_WEAK=true GEOPM_REPORT=geopm.report OMP_PROC_BIND=true GEOPM_PROFILE= LD_PRELOAD=libgeopm.so MV2_ENABLE_AFFINITY=0 KMP_AFFINITY=disabled KMP_WARNINGS=FALSE I_MPI_PIN_DOMAIN=[0x2,0xf0,0xf00,0xf0000,0xf00000] mpiexec.hydra -ppn 5 -n 5 ./print_affinity

Rank: 0 cgroup CPUs: [4,5,6,7] omp CPUs: [6,7,7,5]
Rank: 1 cgroup CPUs: [8,9,10,11] omp CPUs: [8,11,9,10]
Rank: 2 cgroup CPUs: [16,17,18,19] omp CPUs: [16,19,17,18]
Rank: 3 cgroup CPUs: [20,21,22,23] omp CPUs: [20,23,22,20]
```

bgeltz, 2019-06-13:
Ok, there's some sort of issue with IMPI interpreting the CPU mask.  This looks sane to me:

```
$ OMP_NUM_THREADS=4 I_MPI_PIN_DOMAIN=[0x2,0xf0000000000000] mpiexec.hydra -ppn 2 -n 2 ./print_affinity
Rank: 0 cgroup CPUs: [1] omp CPUs: [1,1,1,1]
Rank: 1 cgroup CPUs: [52,53,54,55] omp CPUs: [52,53,54,55]
```

This looks bonkers:
```
$ OMP_NUM_THREADS=4 I_MPI_PIN_DOMAIN=[0x2,0xf00000000000000] mpiexec.hydra -ppn 2 -n 2 ./print_affinity
Rank: 0 cgroup CPUs: [0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,65,66,67,68,69,70,71,72,73,74,75,76,77,78,79,80,81,82,83,84,85,86,87,88,89,90,91,92,93,94,95,96,97,98,99,100,101,102,103,104,105,106,107,108,109,110,111,112,113,114,115,116,117,118,119,120,121,122,123,124,125,126,127,128,129,130,131,132,133,134,135,136,137,138,139,140,141,142,143,144,145,146,147,148,149,150,151,152,153,154,155,156,157,158,159,160,161,162,163,164,165,166,167,168,169,170,171,172,173,174,175,176,177,178,179,180,181,182,183,184,185,186,187,188,189,190,191,192,193,194,195,196,197,198,199,200,201,202,203,204,205,206,207,208,209,210,211,212,213,214,215,216,217,218,219,220,221,222,223,224,225,226,227,228,229,230,231,232,233,234,235,236,237,238,239,240,241,242,243,244,245,246,247,248,249,250,251,252,253,254,255] omp CPUs: [0,1,2,3]
Rank: 1 cgroup CPUs: [0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,65,66,67,68,69,70,71,72,73,74,75,76,77,78,79,80,81,82,83,84,85,86,87,88,89,90,91,92,93,94,95,96,97,98,99,100,101,102,103,104,105,106,107,108,109,110,111,112,113,114,115,116,117,118,119,120,121,122,123,124,125,126,127,128,129,130,131,132,133,134,135,136,137,138,139,140,141,142,143,144,145,146,147,148,149,150,151,152,153,154,155,156,157,158,159,160,161,162,163,164,165,166,167,168,169,170,171,172,173,174,175,176,177,178,179,180,181,182,183,184,185,186,187,188,189,190,191,192,193,194,195,196,197,198,199,200,201,202,203,204,205,206,207,208,209,210,211,212,213,214,215,216,217,218,219,220,221,222,223,224,225,226,227,228,229,230,231,232,233,234,235,236,237,238,239,240,241,242,243,244,245,246,247,248,249,250,251,252,253,254,255] omp CPUs: [0,1,2,3]
IPL WARN> domain mask syntax error, allowed value: hex number list enclosed in []
```
The only delta between the first and second test above is an extra 0 in the second rank's mask.

We never see that `IPL WARN` message when the launcher is used because the Controller ranks are hanging.

bgeltz, 2019-06-13:
In addition to the "0x" issue, `KMP_AFFINITY=disabled` seems to be preventing proper pinning:

This is OK:
```
$ OMP_NUM_THREADS=4 I_MPI_PIN_DOMAIN=[2,f000000000000,f0000000000000,f00000000000000,f000000000000000] mpiexec.hydra -ppn 5 -n 5 ./print_affinity
Rank: 0 cgroup CPUs: [1] omp CPUs: [1,1,1,1]
Rank: 1 cgroup CPUs: [48,49,50,51] omp CPUs: [48,49,50,51]
Rank: 2 cgroup CPUs: [52,53,54,55] omp CPUs: [52,53,54,55]
Rank: 3 cgroup CPUs: [56,57,58,59] omp CPUs: [56,57,58,59]
Rank: 4 cgroup CPUs: [60,61,62,63] omp CPUs: [60,61,62,63]
```

This is not:
```
$ KMP_AFFINITY=disabled OMP_NUM_THREADS=4 I_MPI_PIN_DOMAIN=[2,f000000000000,f0000000000000,f00000000000000,f000000000000000] mpiexec.hydra -ppn 5 -n 5 ./print_affinity
Rank: 0 cgroup CPUs: [1] omp CPUs: [1,1,1,1]
Rank: 1 cgroup CPUs: [48,49,50,51] omp CPUs: [48,51,48,51]
Rank: 2 cgroup CPUs: [52,53,54,55] omp CPUs: [52,54,52,54]
Rank: 3 cgroup CPUs: [56,57,58,59] omp CPUs: [56,59,56,59]
Rank: 4 cgroup CPUs: [60,61,62,63] omp CPUs: [60,63,60,63]
```

This is only an issue on CoolMUC and is not reproducible on internal clusters.

bgeltz, 2019-06-13:
Next, removing `KMP_AFFINITY` makes tutorial_0 work without issue single node:
```
$ GEOPM_TRACE=tutorial_0_trace GEOPM_PROFILE= LD_DYNAMIC_WEAK=true GEOPM_REPORT=tutorial_0_report OMP_PROC_BIND=true GEOPM_CTL=process LD_PRELOAD=libgeopm.so MV2_ENABLE_AFFINITY=0 KMP_WARNINGS=FALSE I_MPI_PIN_DOMAIN=[2,f000000000000,f0000000000000,f00000000000000,f000000000000000] mpiexec.hydra -ppn 5 -n 5 ./tutorial_0
MPI_COMM_WORLD size: 4
Sleeping for five seconds
```
However it still fails multinode (-n is now 10):
```
$ GEOPM_TRACE=tutorial_0_trace GEOPM_PROFILE= LD_DYNAMIC_WEAK=true GEOPM_REPORT=tutorial_0_report OMP_PROC_BIND=true GEOPM_CTL=process LD_PRELOAD=libgeopm.so MV2_ENABLE_AFFINITY=0 KMP_WARNINGS=FALSE I_MPI_PIN_DOMAIN=[2,f000000000000,f0000000000000,f00000000000000,f000000000000000] mpiexec.hydra -ppn 5 -n 10 ./tutorial_0
Warning: <geopm> Controller handshake failed, running without geopm.
<geopm> MPI ranks are not affinitized to distinct CPUs
MPI_COMM_WORLD size: 9
Sleeping for five seconds
Error: <geopm> Runtime error: ControlMessageImp::wait(): mpp3r03c05s01 : is_ctl=1 : is_writer=1 : Timed out waiting for status 2: at geopm/src/ControlMessage.cpp:111
(hang)
```

`print_affinity` is not indicating any issues on 2 nodes:
```
$ OMP_NUM_THREADS=4 I_MPI_PIN_DOMAIN=[2,f000000000000,f0000000000000,f00000000000000,f000000000000000] mpiexec.hydra -ppn 5 -n 10 ./print_affinity
Rank: 0 cgroup CPUs: [1] omp CPUs: [1,1,1,1]
Rank: 1 cgroup CPUs: [48,49,50,51] omp CPUs: [48,49,50,51]
Rank: 2 cgroup CPUs: [52,53,54,55] omp CPUs: [52,53,54,55]
Rank: 3 cgroup CPUs: [56,57,58,59] omp CPUs: [56,57,58,59]
Rank: 4 cgroup CPUs: [60,61,62,63] omp CPUs: [60,61,62,63]
Rank: 5 cgroup CPUs: [1] omp CPUs: [1,1,1,1]
Rank: 6 cgroup CPUs: [48,49,50,51] omp CPUs: [48,49,50,51]
Rank: 7 cgroup CPUs: [52,53,54,55] omp CPUs: [52,53,54,55]
Rank: 8 cgroup CPUs: [56,57,58,59] omp CPUs: [56,57,58,59]
Rank: 9 cgroup CPUs: [60,61,62,63] omp CPUs: [60,61,62,63]
```

bgeltz, 2019-06-13:
Ok after reading:
https://doku.lrz.de/display/PUBLIC/Running+parallel+jobs+on+the+Linux-Cluster

The magic for running multi-node in an interactive session is to set `--tasks-per-node` properly when you issue `salloc`.   I was just now successful running tutorial_0 by issuing:
```
$ salloc -N2 -t 2:00:00 --tasks-per-node=5
(wait for allocation)
$ GEOPM_TRACE=tutorial_0_trace GEOPM_PROFILE= LD_DYNAMIC_WEAK=true GEOPM_REPORT=tutorial_0_report OMP_PROC_BIND=true GEOPM_CTL=process LD_PRELOAD=libgeopm.so MV2_ENABLE_AFFINITY=0 KMP_WARNINGS=FALSE I_MPI_PIN_DOMAIN=[2,f000000000000,f0000000000000,f00000000000000,f000000000000000] mpiexec.hydra -ppn 5 -n 10 ./tutorial_0
MPI_COMM_WORLD size: 8
Sleeping for five seconds
```

I verified that the report contains 2 nodes.

According to that documentation, it looks like specifying `--tasks-per-node` is also required for batch scripts, but I have not yet checked.

bgeltz, 2019-06-14:
If I issue `salloc -N2` then try to use IMPI to control process placement on the nodes, everything gets allocated to the first node:
```
$ mpiexec.hydra -ppn 1 -n 2 hostname
mr-fusion2
mr-fusion2
```

bgeltz, 2019-06-14:
Ok.  This works:

```
$ salloc -N2
$ scontrol show hostnames > hosts
$ mpiexec.hydra -bootstrap slurm -f hosts -n 2 -ppn 1 hostname
srun: Loaded msrsave restore plugin.
srun: Loaded geopm plugin.
mcfly10
mcfly11
```

https://slurm.schedmd.com/mpi_guide.html#intel_mpiexec_hydra
https://scicomp.jlab.org/docs/intelMPIJobs

bgeltz, 2019-06-14:
After doing some more poking around, I'm able to run all the tutorials I've tried so far on CoolMUC3 with `srun` as long as I explicitly set my PMI library prior to invoking the tutorial script:
```
export I_MPI_PMI_LIBRARY=/usr/lib64/libpmi.so
```
(This library doesn't exist on our internal clusters, but it looks like it is available in the OHPC repos as `slurm-libpmi-ohpc.x86_64`.)

If you _have_ to launch via `mpiexec.hydra` you have to specify the hosts you have allocated with `-f <host_file>`.  I'm going to see if I can cook that logic into the launcher somehow.  If SLURM is doing resource management it's easy to get via `scontrol`.  Otherwise it'll have to be manually specified.

bgeltz, 2019-06-19:
Ok.  I think I've gotten to the bottom of all the issues with IMPI, SLURM, and CoolMUC3.

One of the fixes has already been merged to dev:
7777303

The remainder are up for review now:
397916c
2c3f94a

With these patches (currently available on my branch: [bgeltz-issue-614](/bgeltz/geopm/tree/bgeltz-issue-614)), all of the tutorials work by setting `OMP_NUM_THREADS` to something resonable and using `GEOPM_LAUNCHER=impi`.  An example:
```
$ OMP_NUM_THREADS=4 GEOPM_LAUNCHER=impi ./tutorial_0.sh
```

Launching via `srun` also works, but there is a catch:  The `/etc/slurm/slurm.conf` file contains an entry `MpiDefault` that specifies the default type of MPI to be used (more [here](https://slurm.schedmd.com/slurm.conf.html) and [here](https://slurm.schedmd.com/mpi_guide.html)).  CoolMUC3 has this set to `none`.  IMO this is misconfigured and should be set to `pmi2`.  The workaround is to use `--mpi=pmi2` in your `srun` command.  An example:
```
geopmlaunch srun \
          -N ${NUM_NODES} \
          -n ${TOTAL_RANKS} \
          --mpi=pmi2 \
          --geopm-ctl=process \
          --geopm-report=tutorial_2_report \
          --geopm-trace=tutorial_2_trace \
          -- ./tutorial_2
```

commit 8f7f317c (2019-06-24): Add SLURM bootstrap option for IMPI

- Fixes #614.

Change-Id: I96a573bad66253f475a9f4284942f7a43774f050
closed 2019-06-24 by cmcantalupo