geopm/geopm #639: Arbitration functions for controls

dianarg, 2019-06-19:
When writing a control (which is not board scoped) to board domain, the current behavior is to write the same value to every domain.  We want to support evenly splitting the value between all domains for the control (e.g. for power limits).  

This feature will
- Add a new method to the IOGroup interface, requiring IOGroups to provide an arbitration function describing how to convert a larger domain value to the value to write to the native domain
    - e.g. f(control_name: string) -> (std::function(total: double) -> double)
- Have PlatformIO look up the function when writing controls for a larger domain than the native domain of the control.

Open questions:
- How to deal with ratio of power between processor sockets and accelerators.  
-

bgeltz, 2019-06-19:
Related to https://github.com/geopm/geopm/issues/599

Discussion arose because of development on https://github.com/geopm/geopm/issues/303.

dannosliwcd, 2019-06-19:
Related to #641

dianarg, 2020-03-18:
This is really a job for the Agent adjust_platform() method.  It's hard to imagine how it will make sense for individual controls in general.