geopm/geopm #643: geopmwrite error message for invalid domain is not clear

cmcantalupo, 2019-06-20:
If someone were to try to modify the frequency of a CPU on Xeon they might logically try the following command:

    # geopmwrite FREQUENCY cpu 2 1.9e9
    Error: cannot write control: <geopm> Invalid argument:
    PlatformIOImp::push_control(): control name "FREQUENCY" not found: at geopm/src/PlatformIO.cpp:405

This is because the domain of control for FREQUENCY is per core not per cpu, but there is little indication of this from the error message.

cmcantalupo, 2019-06-20:
@crbaird thanks for the feedback.

bgeltz, 2022-03-29:
Seems this is resolved:

```
[bgeltz@mcfly1 geopm](dev)$ geopmwrite CPU_FREQUENCY_CONTROL cpu 2 1.9e9
Error: cannot write control: <geopm> Invalid argument: PlatformIOImp::write_control(): domain 3 is not valid for control "CPU_FREQUENCY_CONTROL": at src/PlatformIO.cpp:642
[bgeltz@mcfly1 geopm](dev)$ geopmwrite -i CPU_FREQUENCY_CONTROL
CPU_FREQUENCY_CONTROL:
    description: Target operating frequency of the CPU based on the control register.
    alias_for: MSR::PERF_CTL:FREQ
    units: hertz
    domain: core
    iogroup: MSRIOGroup
[bgeltz@mcfly1 geopm](dev)$ geopmwrite CPU_FREQUENCY_CONTROL core 2 1.9e9
[bgeltz@mcfly1 geopm](dev)$
```

closed 2022-03-29 by bgeltz