geopm/geopm #647: Implement and test spank plugin accepting a geopm profile string

dianarg, 2019-06-20:
This goal of this prototype is only to test passing --geopm-profile through the commandline args of srun to a spank plugin.  Logic for deciding the appropriate policy based on profile name should be implemented as part of a different story (TBD).

Rough outline of steps:
1. job submitted (jobid)
2. user calls `srun --geopm-profile`
3. spank plugin runs: slurm_spank_init_post_opt(), which can see the --geopm-profile argument value if spank_option struct list is defined.
4. spank plugin uses profile string to determine the best policy
5. spank plugin sends message to geopmd: see #644

dianarg, 2019-07-09:
This is no longer needed because the geopm daemon will handle getting the profile name and agent from the endpoint, and looking up the appropriate policy.  The difficulty the implementation without the daemon is that srun does not know what agent should be used to look up the policy.  It may come from the user or the environment.

A rough initial implementation of the first prototype can be found on my forks:
https://github.com/dianarg/geopm/tree/dianarg-srun-reservation
 https://github.com/dianarg/geopm-slurm/tree/dianarg-policy-spank

dianarg, 2019-07-10:
Reopening to use as proof-of-concept for the near term.  Longer-term the plan is still to implement the daemon.

bgeltz, 2019-08-21:
Please commit to the geopm-slurm Git repo and close this issue.

dianarg, 2019-11-20:
WW45: Code from this plugin is being moved into libgeopmpolicy as part of #786.  #225 is also required to correctly support this plugin running together with the static policy plugin.