geopm/geopm #665: Missing MSR 0x620 in Broadwell whitelist

amarathe84, 2019-07-08:
Due to missing MSR 0x620 in the msr-safe whitelist for Broadwell on Quartz, we're seeing the following error output by GEOPM:

``<geopm> Could not write to MSR device: MSRIOImp::read_msr(): pread() failed at offset 0x620 system error: Permission denied: at geopm/src/MSRIO.cpp:98``

Related to issue #498.

Should this MSR be in the stock msr-safe whitelist? We're preparing for Quartz DAT in a week. We'd like to have this resolved before the DAT.

bgeltz, 2019-07-08:
I just posted LLNL/msr-safe#56 to address this issue.  Some additional background information:

This block of code:
https://github.com/geopm/geopm/blob/d65d25dbccbec2ee6fb1c8eae6ae581fc748009b/src/MSRIOGroup.cpp#L844-L847

Is responsible for loading the MSR description files that GEOPM needs.  BDX and HSX use the same file since they are identical.  GEOPM assume's MSR 0x620 is writable for HSX, BDX, and SKX.

In the PR to msr-safe, I've added this MSR for every supported architecture AFAICT.

cmcantalupo, 2019-07-09:
Note that the commit here:

https://github.com/geopm/geopm/commit/97a850a89a46f9d84f147ee5c1fa607976426a0f

prefixes those messages with "Warning: " which was an oversight in the original implementation.  Though these warnings are not to be ignored if you are using an agent that manipulates the 0x620 MSR (when that write occurs there will be an actual error, not just a warning).  The warning here is to let you know that we cannot restore the value for an MSR that we can neither read nor write.  

That being said, it would be great to add the 0x620 msr to the whitelist with the PR @bgeltz mentioned above.

slabasan, 2019-07-10:
The above mentioned MSR has been merged into msr-safe [here](https://github.com/LLNL/msr-safe/commit/e4c4b0f648c3261e0d0611fe2ffdb9897aff511a)

closed 2019-07-10 by bgeltz