geopm/geopm #687: geopmbench / OMPT : OMPT Regions suppressed  in reports when listed after regular regions

bgeltz, 2019-07-30:
When I run the following config with geopmbench, all is well:
```
{"loop-count": 100,
 "region": ["dgemm-unmarked", "dgemm", "stream-unmarked"],
 "big-o": [0.100000, 0.100000, 0.1]}
```
The report corresponding to the above configuration looks like:
```
##### geopm 1.0.0+dev95g899a7a0 #####                                                                                                               [6/2981]
Start Time: Thu Jan 08 11:17:00 1970
Profile: /p/lscratchh/geltz2/build/geopm/bin/geopmbench
Agent: monitor
Policy: {}

Host: quartz30
Region [OMPT]geopmbench:geopm::StreamModelRegion::run()_40184 (0x000000006ccb2dbe):
    runtime (sec): 18.2092
    sync-runtime (sec): 4.92884
    package-energy (joules): 913.413
    dram-energy (joules): 228.677
    power (watts): 185.32
    frequency (%): 114.799
    frequency (Hz): 2.41079e+09
    mpi-runtime (sec): 0
    count: 100
Region [OMPT]libmkl_intel_thread.so:gemm_omp_driver_v2_2986379 (0x0000000062e155b0):
    runtime (sec): 1.61942
    sync-runtime (sec): 0.0447739
    package-energy (joules): 8.66296
    dram-energy (joules): 1.28595
    power (watts): 193.482
    frequency (%): 114.699
    frequency (Hz): 2.40869e+09
    mpi-runtime (sec): 0
    count: 100
Region dgemm (0x00000000a74bbf35):
    runtime (sec): 1.6153
    sync-runtime (sec): 0.0201255
    package-energy (joules): 4.64417
    dram-energy (joules): 0.699188
    power (watts): 230.76
    frequency (%): 114.637
    frequency (Hz): 2.40737e+09
    mpi-runtime (sec): 0
    count: 100
Region model-init (0x00000000644f9787):
    runtime (sec): 0.449011
    sync-runtime (sec): 0.440202
    package-energy (joules): 77.73
    dram-energy (joules): 14.7723
    power (watts): 176.578
    frequency (%): 123.788
    frequency (Hz): 2.59954e+09
    mpi-runtime (sec): 0
    count: 1
Region unmarked-region (0x00000000725e8066):
    runtime (sec): 0.506853
    sync-runtime (sec): 16.9179
    package-energy (joules): 3324.63
    dram-energy (joules): 773.019
    power (watts): 196.515
    frequency (%): 115.001
    frequency (Hz): 2.41501e+09
    mpi-runtime (sec): 0
    count: 0
Epoch Totals:
    runtime (sec): 21.9002
    sync-runtime (sec): 20.7308
    package-energy (joules): 4029.83
    dram-energy (joules): 955.344
    power (watts): 194.389
    frequency (%): 114.85
    frequency (Hz): 2.41184e+09
    mpi-runtime (sec): 0
    count: 100
    epoch-runtime-ignore (sec): 0
Application Totals:
    runtime (sec): 23.6038
    package-energy (joules): 4491.15
    dram-energy (joules): 1037.03
    power (watts): 190.273
    mpi-runtime (sec): 0
    ignore-time (sec): 0
    geopmctl memory HWM: 152976 kB
    geopmctl network BW (B/sec): 0
```

If I change the bench config to be:
```
{"loop-count": 100,
 "region": ["dgemm", "dgemm-unmarked", "stream-unmarked"],
 "big-o": [0.100000, 0.100000, 0.1]}
```
(That is, I put "dgemm" before "dgemm-unmarked"...) The report looks like:
```
##### geopm 1.0.0+dev95g899a7a0 #####
Start Time: Tue Feb 24 21:32:09 1970
Profile: /p/lscratchh/geltz2/build/geopm/bin/geopmbench
Agent: monitor
Policy: {}

Host: quartz35
Region model-init (0x00000000644f9787):
    runtime (sec): 0.47895
    sync-runtime (sec): 0.470349
    package-energy (joules): 79.6368
    dram-energy (joules): 14.9743
    power (watts): 169.314
    frequency (%): 123.733
    frequency (Hz): 2.5984e+09
    mpi-runtime (sec): 0
    count: 1
Region dgemm (0x00000000a74bbf35):
    runtime (sec): 0.135101
    sync-runtime (sec): 0.129899
    package-energy (joules): 18.8153
    dram-energy (joules): 2.45386
    power (watts): 144.845
    frequency (%): 121.288
    frequency (Hz): 2.54705e+09
    mpi-runtime (sec): 0
    count: 1
Region unmarked-region (0x00000000725e8066):
    runtime (sec): 0.0617347
    sync-runtime (sec): 0.0200873
    package-energy (joules): 2.7417
    dram-energy (joules): 0.414703
    power (watts): 136.489
    frequency (%): 123.365
    frequency (Hz): 2.59067e+09
    mpi-runtime (sec): 0
    count: 0
Epoch Totals:
    runtime (sec): 21.9653
    sync-runtime (sec): 0.144786
    package-energy (joules): 21.0402
    dram-energy (joules): 2.78575
    power (watts): 145.319
    frequency (%): 121.595
    frequency (Hz): 2.55349e+09
    mpi-runtime (sec): 0
    count: 100
    epoch-runtime-ignore (sec): 0
Application Totals:
    runtime (sec): 23.5895
    package-energy (joules): 4409.84
    dram-energy (joules): 1017.46
    power (watts): 186.941
    mpi-runtime (sec): 0
    ignore-time (sec): 0
    geopmctl memory HWM: 152996 kB
    geopmctl network BW (B/sec): 0
```

In the second case, all OMPT regions are absent from the report.

dianarg, 2019-09-10:
I have observed a similar problem: regular regions after OMPT regions suppress all user markup regions from the report.

bgeltz, 2022-07-26:
No longer reproducible.

closed 2022-07-26 by bgeltz