geopm/geopm #691: Document behavior of frequency control

dianarg, 2019-08-07:
Reading the FREQUENCY signal (MSR::PERF_STATUS:FREQ) does not always return the same value as the control (MSR::PERF_CTL:FREQ).  This is surprising and we have gotten questions from multiple people about this issue.  The main use case is to confirm that an application running at a fixed frequency is really running at that frequency.  This assumes RAPL is not lowering the frequency. There are a few ways to do this: 1) as long as there is sufficient load on the processor, sampling PERF_STATUS should be reasonable 2) use the ratio of clocks unhalted ref to thread as in the Reporter to estimate achieved frequency.

dianarg, 2019-08-07:
An example script:
```
##
#!/bin/bash
#SBATCH -N 1
set -x

sticker_freq=$(srun geopmread CPUINFO::FREQ_STICKER board 0)

# write desired frequency to all cores; writing sticker disables turbo
srun geopmwrite FREQUENCY board 0 $sticker_freq

# sample FIXED_CTR2
start_ref=$(srun geopmread CYCLES_REFERENCE board 0)
# sample FIXED_CTR1
start_thread=$(srun geopmread CYCLES_REFERENCE board 0)

# Run the application here
#./run_app
# example high CPU command
find /usr > /dev/null

end_ref=$(srun geopmread CYCLES_REFERENCE board 0)
end_thread=$(srun geopmread CYCLES_REFERENCE board 0)

python -c "print(($end_ref - $start_ref) / ($end_thread - $start_thread) * $sticker_freq)"
##
```

dianarg, 2019-08-07:
Put in README, maybe also in control description from `geopmwrite --info`

dianarg, 2019-09-04:
Might want to put in man page, have README point to man page.

dianarg, 2019-11-26:
Related: #795

bgeltz, 2022-03-29:
The ```FREQUENCY``` alias has been removed.

closed 2022-03-29 by bgeltz