geopm/geopm #692: Look into GROMACS

dianarg, 2019-08-07:
Test with GEOPM.  Tune problem size until we have a run that is both realistic and shows benefits with a GEOPM agent (prefer EE agent).

Might want more issues e.g.:
- Add region markup to GROMACS
- Find correct problem size
etc.

bgeltz, 2019-08-21:
Follow up with LRZ again.

bgeltz, 2019-08-21:
The best info I've found to date on how to tune GROMACS is here:
https://repository.prace-ri.eu/git/UEABS/ueabs/blob/master/gromacs/README_ACC.md
    
I've been running Test Case A:
https://repository.prace-ri.eu/ueabs/GROMACS/1.2/GROMACS_TestCaseA.tar.gz

The best performing config for me was:
88 ranks per node
1 OMP Thread
48 PME only ranks
    
Options to GROMACS: -dlb yes -npme 40 -dd 8 3 2

This has uncovered a few new issues:
1. #697 
2. The fix for 697 does not work for GROMACS.  I suspect this has to do with how GROMACS splits up the per-rank work, and which ranks make the API call to ```geopm_prof_epoch()``` , but more debug is necessary.  **TODO** : File a new issue once the problem is characterized.
3. The launcher chokes on the following setups:
    a. 88 ranks per node, 1 OMP thread (the same as what I listed above as my best performer)
    b. Any number of ranks per node above 40 on our dual socket 22-core cluster.

Debug is ongoing.

bgeltz, 2019-08-23:
> 2\. The fix for 697 does not work for GROMACS.  I suspect this has to do with how GROMACS splits up the per-rank work, and which ranks make the API call to `geopm_prof_epoch()` , but more debug is necessary.  **TODO** : File a new issue once the problem is characterized.

My suspicion ended up being correct.  The ranks that were split off for the PME calculations have their own processing loop which also requires an ```epoch()``` call.  I now have valid reports.

bgeltz, 2019-11-20:
Aside from minor tuning options during build time, I have no new datasets beyond the ones I've previously identified on the UEABS site.

cmcantalupo, 2019-12-03:
Relates to #809